Ultimaker/Cura #9309: Part of complex model do not get created

nicwilson58, 2021-02-22:
**Application version**
(The version of the application this issue occurs with.)
4.8.0

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows 10 latest version and patches.
**Printer**
(Which printer was selected in Cura?)
CR10S Pro
**Reproduction steps**
1.  Load model
2.  project included below
3.  std settings,  only change res to 0,16
4.  look at preview
5. preview missing an area as shown
6. model prints same as preview with missing part

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
Prepare view:
[image]
Preview view of same area,  upper arch of 3 missing:
[image]

**Actual results**
(What happens after the above steps have been followed.)
model prints as the preview does with the upper arch missing

**Expected results**
(What should happen after the above steps have been followed.)
Prepare and Preview should be the same,   sliced model with everything included.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

https://www.dropbox.com/s/exs4a76na82vr3z/CCR10SPRO_HarryPotter_Hogwarts%2BPlate_Repaired.3mf?dl=0

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
N/A  you can generate the same issue as me just by slicing
**Additional information**
(Extra information relevant to the issue.)

Asterchades, 2021-02-22:
If you've already excluded the possibility of the model have incorrect normals (which the "repaired" part of the name suggests, but it's worth double-checking) then odds are those parts of the model are just too thin. They look like they'd be at most one extrusion wide, but there's a very good chance they're not even that.

The Arachne alpha is designed to accommodate for things like this. Obviously it's only an alpha and has other issues of its own so that's not the immediate solution, but it does mean in future it might not be a problem.

You could also try the **Print Thin Walls** option but that may result in a less-than-desirable outcome due to how the present versions actually accomplish that.

It could be worth checking with a very small amount of **Horizontal Expansion** on the shell. 0.1mm to 0.2mm tops. It may remove some of the more fine details for things that indent, which is why you don't want too much, but could "fatten up" the thin parts enough to make them sliceable. Scaling the model up would have a similar effect, but may not be an option depending on the size of your printer.

fvrmr, 2021-02-23:
Hi @nicwilson58 thank you for your report. I looked into your model and I really believe it has something to do with your model. Also what @Asterchades is suggesting is correct. Your walls are probably to thin. 

I sliced your model in Arachne and there the walls are printed. (I cut off a part to slice it faster)
[image]

You can read here more about Arachne: https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_alpha

closed 2021-03-09 by no-response[bot]