Ultimaker/Cura #9310: [4.8.0] Tree support not fully generated at overhang

fyrcraeft, 2021-02-22:
**Application version**
4.8.0

**Platform**
Windows

**Printer**
Probably all

**Reproduction steps**
1. Add model with overhang
2. Activate Tree Support (reducing Support Overhang Angle to 5 to make it more visible)
3. Slice

**Screenshot(s)**
[image]
[image]

**Actual results**
Overhangs with around 55° and less will not be (fully) supported

**Expected results**
All surfaces should be supported

**Project file**
[UMS5_overhang_test.zip](https://github.com/Ultimaker/Cura/files/6021577/UMS5_overhang_test.zip)

**Log file**
Shouldn't matter

**Additional information**
I just added a Ultimaker S5 (I just have a custom made printer, but result is the same), didn't change anything beside support settings. Could also be an Engine topic since normal support seems to work.

It is working with normal support
[image]

Support XY distance settings seems to have an impact here
[image]

fvrmr, 2021-02-23:
Hi @fyrcraeft thank you for your report. 
This is not a bug, but expected behaviour. 

Because the Z becomes more important when the angle is getting less steep.
And that's why you will see when you change the Support X/Y Distance the support is printed. 

Please let me know if this makes sense to you.

fyrcraeft, 2021-02-24:
Hi @fvrmr,
thanks for your answer.
To be honest, I don't think it makes sense to have missing support when it's needed. But when this is how Cura is handling it, then it is like it is. I don't really want to start a discussion about it.

closed 2021-02-24 by fyrcraeft

Ghostkeeper, 2021-03-02:
Actually, the tree support is expected to support the same things as the normal support. The X/Y and Z distance influence the distance that the support should keep, but shouldn't cause the tree support to completely disappear.

I think you're hitting the bug discussed here: https://github.com/Ultimaker/Cura/issues/8614
If you want to add your point of view to that discussion it could help, but there are already a couple of project files there and in other tickets linking to it. I definitely think it's a bug.