Ultimaker/Cura #9312: 4.8 is missing an MSI version on the downloads.

evertoaster, 2021-02-22:
I work at a school and we want to deploy cura to all our classroom pcs and until recently this was easily done with group policy deploying the cura msi.  4.8 doesn't have a msi just an exe.

Please can you create an msi version for download.

fvrmr, 2021-02-22:
Hi @evertoaster thank you for your feature request. 
There is indeed no msi for 4.8. The reason for this is because we have now Ultimaker Essentials.
If you want a msi you have to sign up for Essentials. You can read more about Essentials here: https://ultimaker.com/software/ultimaker-essentials

That's why I will label this with won't do/fix. Hope you understand.

evertoaster, 2021-02-23:
Lame

closed 2021-02-23 by nallath