Ultimaker/Cura #9314: Buildplate intersection not clearly marked

sjoerdtimmer, 2021-02-22:
**Application version**
Cura 4.8.0

**Reproduction steps**
Open stl
Move stl down into the buildplate partially

As noted in #9308 there is at least one way to unintentionally move objects into the buildplate without noticing; making this less of a contrived situation than it may appear. 

**Actual results**
part of the model that is intersecting the buildplate is shown in a darker shade of blue, but this is obscured by the build-plate itself when looking from above. The fact that part of the model will not be printed can only be seen when looking from the bottom or from a low angle (depending of course on how far the model extends under the buildplate).

In addition, it's only visible in the Prepare tab and disappears in the Preview tab.

**Expected results**
Clearly shown that part of the model will not be printed due to it intersecting with the buildplate.

Maybe some shader trick can be used (like in xray mode) to bring the problematic area into view also when it's below the buildplate. In addition, the dark blue color is not very striking, a more warningy color like red or orange might be more appropriate.

Alternatively show a pop-up warning message if this happens

** Extra info **
The issue is somewhat alleviated by the fact that the lightblue outline also turns a little darker around "submerged" edge but this is not very noticeable. In particular in the preview tab it's easy to miss.

fvrmr, 2021-02-23:
Hi @sjoerdtimmer thank you for your report. I will change this to a feature request and then discuss this with the team. 
Keep you posted.

fvrmr, 2021-02-26:
Hi @sjoerdtimmer I have discussed it with the team. And I have to reject this feature request. Because it is hard to implement this. If we want this we need rebuild to the entire graphical interface.
Hope you understand.

closed 2021-02-26 by fvrmr