Ultimaker/Cura #9315: Optimize in-plane orientation of prints

luotinen, 2021-02-23:
The in-plane orientation of the print will influence the print time. It is not intuitive which print direction will minimize the print time while maximizing the print quality. I suggest a separate button or function "Optimize XY orientation". It would simply loop through the in-plane angles and make note of the estimated print time (and any quality factors) and then pick the optimal direction. This would help users shave off some print time, save energy, and potentially a tiny bit better print quality.

fvrmr, 2021-02-23:
Hi @luotinen thank you for your feature request.
Could you please fill in the form that helps us understand your feature request better.
Thank you in advance.

luotinen, 2021-02-23:
Sure!

**Is your feature request related to a problem? Please describe.**
The in-plane orientation of the print will influence the print time. It is not intuitive which print direction will minimize the print time while maximizing the print quality.

**Describe the solution you'd like**
I suggest a separate button or function "Optimize XY orientation". It would simply loop through the in-plane angles and make note of the estimated print time (and any quality factors) and then pick the optimal direction.

**Describe alternatives you've considered**
The current alternative is to manually try out the different orientations and pick the best. This is time consuming.

**Affected users and/or printers**
This feature would benefit all users and it is applicable on all printers.

**Additional context**
This would help users shave off some print time, save energy, and potentially a tiny bit better print quality.

StuSerious, 2021-02-23:
To understand better.. what you're suggesting is that lines that are oriented like this:
[image]
Be oriented like this:
[image]
Or are you suggesting that the model itself be re-oriented?
___
I can see your point but I'm wondering how well it actually scales with more complex models, for example:
[image]
[image]
Re-orienting all lines simply 90° "only" shaves off 2.53485% of the print time. Yes this is a time gain but I'd imagine it would be even smaller with smaller models. (and most models won't have 100% infill)
Engine-wise I don't know how feasible it would be..
If I recall correctly Cura computes layers in parallel, so these calculations would add significant overhead and complexity for a relatively small time gain. (devs please correct me)

luotinen, 2021-02-23:
Actually, there is no computational cost if it would only be triggered by the user when requested. For example, right-click the model and select "Optimize XY orientation".

On my model, I am getting a significant time saving, but I tried some others and the benefit is smaller then.

I also spotted there is an Auto Orientation extension by Nallath but is does not seem to consider XY-rotation.

Ghostkeeper, 2021-03-02:
It's important to know that there are two effects here:
* Cartesian printers can accelerate faster in the diagonal X/Y direction because they will use both motors to accelerate then. Marlin models this and so does Cura's time estimation algorithm. This is why by default, most skin and infill lines are oriented diagonally.
* Depending on the shape of a model, an area might be filled with more or fewer lines by orienting those lines differently w.r.t. the rotation of the model.

Both of these effects can be optimised by manually specifying the directions of those lines with the "Infill Line Directions" and "Top/Bottom Line Directions" settings.

Optimising this automatically should indeed be done in the front-end similar to the Auto Orientation plug-in (which only optimises for support), because it significantly alters the position of the model and the user needs to know that, especially if there are multiple objects on the build plate that it might collide with.

I think it would be pretty hard to optimise for this though! Perhaps some heuristic can be thought of that can be calculated in a reasonable amount of time. Like... Multiply all of the normal vectors of all triangles of the model by the area of those triangles, and then sum the normal vectors to get an "average normal vector" of the whole model's surface. And then rotate the object such that the average normal vector is in a diagonal of the X/Y plane?