Ultimaker/Cura #9316: Cura not start to slicing the object.

Niassamond, 2021-02-23:
Im trying to slice a litophane, it has 150mm height by 100mm width, whenever i click slice the program appears as slicing, but the program itself never starts. It doesnt even try to slice the file it just hangs in there waiting saying its slicing forever. i waited for 10 minutes and nothing. My PC is Windows 10, Nvidia 1050Ti, 16GB Ram. Any help is appreciated. i Already sliced 3 day prints sucessfully and i dont think a small litophane like this will take more than that.

StuSerious, 2021-02-23:
@Niassamond could you actually fill out the form or at the very least provide a Project file so people can understand the issue better and try to replicate the Issue?
Thanks! <3

Niassamond, 2021-02-23:
> @Niassamond could you actually fill out the form or at the very least provide a Project file so people can understand the issue better and try to replicate it?
> Thanks! <3

Hello, sorry i never used github and im a bit noob at this, how do i post the STL file? and also where is the form file?

StuSerious, 2021-02-23:
No worries! 
And the contributing guidelines here: https://github.com/Ultimaker/Cura/blob/master/contributing.md
You can find the template here: https://github.com/Ultimaker/Cura/blob/master/.github/ISSUE_TEMPLATE/bug-report.md
To attach a files that aren't supported by GH you could just .zip them up or use a service like WeTransfer!

Ghostkeeper, 2021-03-03:
The project file includes the 3D model you're slicing, so if that 3D model contains a private picture you might not want to upload it publically. You can also send it exclusively to me (or any of the other developers). My work e-mail is `r dot dulek at ultimaker dot com`.

Unfortunately if CuraEngine crashes, the front-end doesn't always pick up on that. Usually it shows the progress bar empty then, but sometimes the progress bar still seems to freeze. You can tell the difference then in your task manager if you want to be really sure, because if it's just very slow it'll still use 100% CPU. If the engine crashes it's definitely a serious bug.

closed 2021-03-09 by no-response[bot]