Ultimaker/Cura #9317: Pause at layer or height not working. will pause for only 5 seconds then restart itself

debie3366, 2021-02-23:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

GregValiant, 2021-02-24:
If you don't fill out the form there is essentially zero information to go on.  If this is happening on a  Creality printer with a 4.2.X mainboard, it's a Creality firmware problem.

debie3366, 2021-02-24:
Hi sorry about that, I have a creality 3d pro. the firmware is Marlin 1.01 the motherboard is 4.2.2 
Kind regards Debbie

fvrmr, 2021-02-25:
Hi @debie3366 as @GregValiant mentioned this is probably a firmware issue. 
You can read more here: https://github.com/Ultimaker/Cura/issues/9272#issuecomment-780923297
Where @GregValiant is suggesting a fix. You could try that. 

Please let us know if that works for you in that thread, so I can close this one.

closed 2021-02-25 by fvrmr