Ultimaker/Cura #9318: Suggestion: "Z hop when retracted" linked sub option "Only Zhop after Initial Bottom Layers"

tastyratz, 2021-02-23:
**Is your feature request related to a problem? Please describe.**
When using Z hop when retracted and gooey stringy filaments like PETG an immediate zhop after extrusion can cause the filament to pull away from the build plate.

**Describe the solution you'd like**
Z hop when retracted can be a great way to prevent hitting the model, but, risk of collision happens higher in the print after things have cooled.
What if there was an option to only start doing a z hop after x layer with the default value linked to "Initial bottom Layers" (but otherwise overrideable)?

This could help ensure a better initial or lower layer adhesion, improve print time, and allow later zhops when parts collide.

fvrmr, 2021-02-25:
Hi @tastyratz thank you for your feature request. I will discuss this with the team. Keep you posted.

fvrmr, 2021-02-26:
Hi @tastyratz I have discussed it with the team and we have decided to defer this. 
The reason why we defer this is because we believe that the print speed would be minimal. And with a cutting mesh you can change the Z hop to a layer you want. 
What you basically ask is layer settings and that would be a big change in our architecture.

Hope you understand. 
Please reach out if you have questions.

closed 2023-07-13 by github-actions[bot]