Ultimaker/Cura #9319: the profiles disappear with my second ultimaker, doesn't matter how many times i reinstall the program, not installed is the warning

wallacecme, 2021-02-23:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

wallacecme, 2021-02-23:
operating with big sur operating system and s5 no 2

fvrmr, 2021-02-25:
Hi @wallacecme could you please fill in the form and add a log file and project file.
Thank you in advance.

Ghostkeeper, 2021-03-03:
The project file is probably unnecessary here since it's not a slicing problem as far as I can see. But the log file is definitely very important and what printers/profiles you're using!

closed 2021-03-11 by no-response[bot]