Ultimaker/Cura #9320: the labels for the icons on the left have ceased to operate, I have uninstalled and deketed the remaining folders and re-installed but this has not fixed the issue

andr001967, 2021-02-24:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fvrmr, 2021-02-25:
Hi @andr001967 could you please fill in the form and add a log file and project file.
Thank you in advance.

Ghostkeeper, 2021-03-03:
Duplicate of https://github.com/Ultimaker/Cura/issues/8324. It's fixed for the next release.

closed 2021-03-04 by Ghostkeeper