Ultimaker/Cura #9321: my view has become corrupted, the build plate angle has changed to something I am sure would not be intentional

andr001967, 2021-02-24:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

andr001967, 2021-02-24:
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/6032472/DxDiag.txt)

andr001967, 2021-02-24:
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/6032474/DxDiag.txt)

fvrmr, 2021-02-25:
Hi @andr001967 could you also fill in this form. And add you log file and some screenshots.
Thank you in advance.

Ghostkeeper, 2021-03-03:
If the camera got positioned strangely (which can happen sometimes when "Center camera when item is selected" or "Zoom toward mouse direction" is enabled) there are these buttons in the bottom left that will reset your camera position and orientation:

[image]

closed 2021-03-11 by no-response[bot]