Ultimaker/Cura #9322: Cura won't slice when assigning stls to Extruder 2 and 3 and disabling Extruder 1 (T0)

Soureaver666, 2021-02-24:
**Application version**
(The version of the application this issue occurs with.)
Cura 4.8.0

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows 10 64bit 1909

**Printer**
(Which printer was selected in Cura?)
Custom FFF Printer with 3 tools

**Reproduction steps**
1. set support and buildplate adhesion extruder to extruder 2
2. disable extruder 1
3. load both bi-color benchy .stl files from https://www.thingiverse.com/thing:763622 into Cura
4. assign extruder 2 to one .stl (should be already predefined here) and extruder 3 to the other one
5. merge models

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
merged model gets shaded and becomes "unable to slice"

when enabling extruder 1 (T0) again -> slicing works

I guess this behavior results from automatically inserting and also pre-heating "T0" when pre-heating the print bed at the start of the print. 
[image]

**Expected results**
model should be slice-able with extruder 1 disabled

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fvrmr, 2021-02-25:
Hi @Soureaver666 thank you for your bug report. 
I can reproduce this to. Will discuss this with the team. Keep you posted.

fvrmr, 2021-02-26:
We have discussed it with the team. And can also reproduce this with a dual extrusion, we've created a ticket on our backlog.
Developers see CURA-8054

closed 2021-11-05 by nallath