Ultimaker/Cura #9323: [4.8] Support Join Distance makes support structures start in thin air.

printingotb, 2021-02-24:
**Application version**
4.8

**Printer**
Custom fff

**Reproduction steps**
1. Download my project and change the Support join Distance to for example 1mm or 3mm

**Screenshot(s)**
[image]
[image]

**Project file**
[Support join distance.zip](https://github.com/Ultimaker/Cura/files/6035920/Support.join.distance.zip)

fvrmr, 2021-02-25:
Hi @printingotb thank your for your report.
I can't reproduce this because your project file doesn't contain the printer and profile settings. It's just a 3mf file. 
Could you please upload a new project file?

printingotb, 2021-02-26:
Oops, here it is:
[Support join distance.zip](https://github.com/Ultimaker/Cura/files/6049608/Support.join.distance.zip)

Ghostkeeper, 2021-03-03:
The issue doesn't reproduce directly with that project file in the current master, but still happens by increasing the Support Join Distance to 5mm. Also should mention that the "Automatically drop items to build plate" preference needs to be disabled when loading that project file in, or the Benchy will sink.

This problem is a (rather extreme) case of the weakness of the zigzag pattern. The zigzag pattern connects the lines on only one side of the support structure. Once the two pieces of support join together, in this case the old piece happens to fall entirely within the space of two lines where the connecting piece happens to be on the other side.
[image]

I'm not sure what we can really do about it. If you think in the direction of trying to guarantee that all lines are supported by something, you'll very quickly end up with a result that is just equal to an extra support wall, defeating the purpose of the zigzag pattern. Using an extra support wall is a workaround. Or increasing the density of the support a bit so that there is a line in that space that supports the extra piece.

printingotb, 2021-03-03:
I was actually printing this benchy to test support settings among other things. I didn't see the issue until it was printed and the support structure had failed.
It sounds like a hard nut to crack. The first that comes to mind is extending every air support line down to the buildplate but that could add a lot of extra support structures? Would it be possible to have cura detect the issue and print an extra wall only where needed?

If I disable the interface, that support structure rests entirely in air:
[image]

On the other hand, the lines pattern doesn't print any support in those areas with this low density. 
Using 10% zigzag and one gradual support infill step gives basically the same result as 5% and no steps.
Enabling support towers with a high enough diameter helps but towers tend to add very thin pillars to places where normal support doesn't even try to reach.

Ghostkeeper, 2021-03-04:
Basically what happens here is that you get a support area like this on the higher layers:
[image]

And then a support area like this on the lower layers:
[image]

For the individual layers the zigzag pattern is doing its work correctly, and these are filled asynchronously. But the lines on one side are not supported on the other because of the zigzag pattern.

printingotb, 2021-03-08:
Thanks, I'll simply tweak my settings with this in mind.

Ghostkeeper, 2021-03-11:
Sorry, I'd prefer to keep this issue open. It's not something we can solve immediately but it's good to know that it exists, and maybe there is suddenly a brilliant idea somewhere to improve this scenario.

printingotb, 2021-03-11:
Oh that's great 🙂

printingotb, 2023-06-11:
Behavior is same/similar although I had to reduce Support overhang angle to get the support structures on the side in Cura 5.3.1

MariMakes, 2023-07-10:
Hey @printingotb,

Sorry, it took us a while to get back to you 😞 

I took a look at the issue, we did introduce a lot of changes since 4.8 related to support. 
I can reproduce your issue after tweaking the angle. 
[image]
I also checked and some printers have the Support Join Distance = 3, by default. 

I'll bring it up to the team to see what they can do to improve it. 
It does however seem to be super model specific. I expect that if it's not a quick fix that it will not get a high priority to implement. Fingers crossed 🤞

closed 2023-07-18 by MariMakes