Ultimaker/Cura #9324: Exclude the extra skin walls from "Small feature speed"

printingotb, 2021-02-24:
**Is your feature request related to a problem? Please describe.**
Enabling small feature speed is great for printing small holes and threads and minis but a side-effect is that expanded skin is reduced in speed as well, increasing printing time dramatically for certain geometries. 

**Describe the solution you'd like**
Make the setting only affect inner and outer walls.

**Affected users and/or printers**
Anyone

**Additional context**
[image]

fvrmr, 2021-02-25:
Hi @printingotb thank you for your feature request.
I will discuss this with the team. Keep you posted!

fvrmr, 2021-02-26:
We have discussed it and this feature request is on our backlog now. 
Developers see CURA-8052.

printingotb, 2021-02-26:
Thanks a lot 🙂

fvrmr, 2021-05-21:
Closing this issue because it has been removed from the backlog. We understand the need of it, but don't see anytime to work on this anytime soon.

fvrmr, 2021-06-04:
Changing the status to defer instead of won't fix/do because we do not have the time to work on it, some one else is always welcome to work on this and create a pull request.

closed 2023-07-13 by github-actions[bot]