Ultimaker/Cura #9325: AppImage is missing .desktop integration.

leoheck, 2021-02-24:
AppImage is missing .desktop integration.

jellespijker, 2021-02-26:
Hi we seriously need additional information here. Please fill in the bug report.

jellespijker, 2021-02-26:
Assuming your are filling issue is with 4.8 then it's now fixed with this PR #9229

leoheck, 2021-02-26:
I am sorry. 

The bug report is for any Cura <= 4.8
This app is not being installed as the other appimages I find on the internet.
It does not create a `.desktop, file for instance.

[AppImageLauncher](https://github.com/TheAssassin/AppImageLauncher) does not recognize it too, to move it to a known folder. This Appimage is like an incomplete job.

So, the fix is not for the 4.8, is it? 
I downloaded it recently, last week and this `.desktop` file is still missing there.

Maybe the fix is going to be in version 4.9 that was not released yet.

jellespijker, 2021-02-27:
You're correct, this fix is not for 4.8.
We can't fix versions that we already released. This should however be fixed for the next version 4.9
I will therefor close this ticket.

closed 2021-02-27 by jellespijker

leoheck, 2021-04-25:
@jellespijker Cura 4.9 was released, there is no desktop integration, yet. What has happened? Bug? Am I doing something wrong?

jellespijker, 2021-04-25:
Do you mean that the AppImage doesn't create a .desktop file, which you can access from your start menu or from your desktop. Just like a Windows or Mac computer?

That is because an AppImage isn't an installer. Normally you download the AppImage give it executable rights and put it in a location which is in your path or accessible by you.

Now if you want to run such a program and start it from your menu or from your desktop you have to create a .desktop file for this. There are numerous tutorials online which tells you how to do this for your distro.

I personally have a KDE based desktop. I put my images in a directory e.q. /home/username/bins/cura or something similar.
And then I right click the AppImage in Dolphin (file explorer) select is executable. I then right click the start menu button and select add new menu entry. From here I create a new menu entry which points towards the AppImage.