Ultimaker/Cura #9327: Profiles disappear

monarrezcr, 2021-02-25:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. Opened Cura
2. Imported profile
3. When making material change, profile changes to 'Not Supported'
4. This has been replicated using different profiles from different sources including local changes to default profiles on different computers.
5. Can fix by trying again and not making changes to anything, then closing Cura and reopening. May happen again if changes to materials are made but not settings.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fvrmr, 2021-02-25:
Hi @monarrezcr could please give us more information, like your version, OS, screenshots and a log file would be helpful. 
Thank you in advance.

Ghostkeeper, 2021-03-04:
I think the profile is just based on a quality level that is not supported by the new material you switched out? What printer are you using and what nozzle and materials? What quality level is the custom profile based on?

closed 2021-03-11 by no-response[bot]