Ultimaker/Cura #9329: cannot rotate object under MacOS BigSur 11.2.1 - impossible to select the circle to turn the object

pljda, 2021-02-25:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fvrmr, 2021-02-25:
Hi @pljda thank you for your report.
This is a duplicate of #8745 we hope to fix this soon. That's why I will close this issue. 
You can read on [our website](https://support.ultimaker.com/hc/en-us/articles/360018141480-Ultimaker-Cura-won-t-run-on-my-new-Macbook) more about this. 

And for next time, please fill in the form instead of just a title, without any information we can't really help you.

closed 2021-02-25 by fvrmr

pljda, 2021-02-25:
Hi, thanks for your comment.
I have relaunched the app and can select the item & rotate.
Strange !
Anyway - youra pp is great - thanks