Ultimaker/Cura #9330: print preview stopped showing

JJ460, 2021-02-25:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

JJ460, 2021-02-25:
cura 4.8,  I changed the initial line width and the print preview display changed to all grey and would not execute the print preview.

Ghostkeeper, 2021-03-04:
A project file and log file would help us reproduce the issue. I've changed the Initial Layer Line Width to 200% but the layer view still shows fine for me.

closed 2021-03-18 by no-response[bot]