Ultimaker/Cura #9333: Outer wall offset insets everything when using only one perimeter

whosawhatsis, 2021-02-26:
**Application version**
4.8

**Platform**
MacOS X

**Printer**
Any (Ultimaker 3 is shown)

**Reproduction steps**
1. Set "Wall line count" to 1
2. Set "Outer Wall Offset" to a positive value (values greater than one make the problem more obvious).

**Screenshot(s)**
[image]

**Actual results**
Everything is inset horizontally, as if a negative value for Horizontal Expansion had been used.

**Expected results**
The (single) perimeter should be inset so that it overlaps with the infill (or with the "Extra infill walls", if used).

**Project file**
[UM3_threadblock.3mf.zip](https://github.com/Ultimaker/Cura/files/6047230/UM3_threadblock.3mf.zip)

**Log file**

**Additional information**

whosawhatsis, 2021-02-26:
For context, what I'm trying to do is print my skin with a single perimeter, and two perimeters around my infill, with the second one provided by the "Extra Infill Wall Count" setting, but I want an overlap between the inner (extra infill) wall and the outer wall. None of the available inset/overlap settings will make these two walls overlap one another.

fvrmr, 2021-02-26:
Hi @whosawhatsis thank you for your bug report. This is fixed by our Arachne release.
[image]

You can read more about Arachne [here](https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_alpha). 
I will close this issue and marked as solved/fixed. But please reach out if you have questions.

closed 2021-02-26 by fvrmr

Ghostkeeper, 2021-03-04:
That's pretty interesting. I wouldn't expect that this reproduces the skeletal graph if you set the inset real big. It's not properly insetting in the tips of the shape then, is it?