Ultimaker/Cura #9334: Importing Infill Patern of Honeycomb/Hexagonal Strucrure

motuhin, 2021-02-26:
**Main message:**
I contacted Ultimaker North American support center to help me import the honeycomb/hexagonal structure as the infill pattern using Cura software. They recommended that I submit a request here in GitHub, and so here I am.
I am aware that the current list of patterns includes Tri-hexagonal structures. However, I am looking for a Hexagonal/Honeycomb without the triangle for creating a research design.

Also, in general, I am wondering how can I possibly import the external infill patterns that are not already included in the list.

Thank you and regards,
Mohammad Tuhin,
University of Houston (TX, USA)

**Application version**
Ultimaker Cura 4.8.0

**Platform**
Windows

**Printer**
Ultimaker S3

**Reproduction steps**
N/A

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

smartavionics, 2021-02-26:
For a Cura variant that provides a honeycomb infill pattern see https://github.com/smartavionics/Cura/releases

Ghostkeeper, 2021-02-26:
We already have an open feature request for a honeycomb pattern here: https://github.com/Ultimaker/Cura/issues/2069
Please refer to that issue instead.

As for "importing" a custom pattern, what you can do is generate a 3D model consisting of the shapes you like (as loose triangles/planes) and then setting that model to be printed in Surface Mode. It's not ideal with the planning of lines, but it'll print an arbitrary shape.

closed 2021-02-26 by Ghostkeeper