Ultimaker/Cura #9338: Change angle of first layer after supporty

loggyboy, 2021-02-27:
My first layer after a support structure is a fixed angle (appears to be about 10 degrees). I cannot find any setting to adjust the angle, or pattern that this initial layer is. General shell settings do not affect this one layer.
Layer 72, shows last layer of support. (concentric)

[image]

Layer 73 shows first layer of shell after support - I cannot find a way to adjust this layer.

[image]

Layer 74 shows 2nd layer of shell, this is affected by shell setting of direction and pattern. (lines, 45deg)

[image]

fvrmr, 2021-03-01:
Hi @loggyboy thank you for your feature request. Could you provide us more information, why is this better and what will affect this and a project file, so I can take a closer look.

Ghostkeeper, 2021-03-04:
Looks like it's just bridging. Adjusting the Bridge Skin Support Threshold should allow you to tune it better.

loggyboy, 2021-03-04:
> Looks like it's just bridging. Adjusting the Bridge Skin Support Threshold should allow you to tune it better.

Bridge was switched off (was only in experimental so hadnt looked at this). However turning this on has now aligned the first layer with the others so I will play around with these settings. Thanks.

fvrmr, 2021-03-05:
I will close this issue, since this is solved by enabling the bridge settings. 
But please feel free to reach out if you have more questions.

closed 2021-03-05 by fvrmr

Ghostkeeper, 2021-03-08:
Counterintuitively, bridging can only be turned off by enabling the setting to get more bridging settings, then setting those bridging settings such that bridging is effectively turned off.