Ultimaker/Cura #9339: [4.8] With 1 or 2 Perimeters and 0 infill, Errant Travel Move

CCS86, 2021-02-27:
4.8.0 / Win 10 x64

When making a print with 1 or 2 perimeter walls and 0% infill, Cura is inserting a small travel move inwards. With 3 walls, the behavior stops:

[image]

[image]

[image]

[UMO_Circle.3mf.txt](https://github.com/Ultimaker/Cura/files/6053696/UMO_Circle.3mf.txt)

Asterchades, 2021-02-27:
Dark blue is a travel without retraction. I would be suspecting that's an Outer Wall Wipe, which would appear to be set to a (in my opinion excessive) level roughly equal to twice the wall width.

The behaviour also isn't stopping with three walls. It's just not moving far enough to cross over the full width of the wall, but it's quite clearly still performing the same movement.

CCS86, 2021-02-27:
The alternate theory sounds correct. Since outer wall wipe distance was set to zero. 

Here is the behavior with a 1mm wipe which pushes the position of this inwards dive:
[image]
[image]
[image]

CCS86, 2021-02-27:
Finally found a setting that changes the behavior: combing mode

Off: 
[image]

All: 
[image]

Not in Skin: 
[image]

Within infill: 
[image]

Asterchades, 2021-02-27:
Ah, combing! Would never have even guessed that it could be at fault here, so nicely deduced.

If I'm honest, I've had it turned off for a long time now. PETG in particular (which I almost exclusively print these days) shows noticeable quality loss with it in use - and situations like this could be part of why. But that's starting to get off topic, so I'll leave it there and hope that the devs can get you sorted.

CCS86, 2021-02-27:
I tried a LOT of settings before that! 

I have gone back and forth on combing. With PLA, using combing can have a pretty dramatic reduction in retracted moves, which on a bowden printer can save you a bunch of time and improve quality. 

Hopefully, the devs can dig in and see why combing is adding these moves. Maybe it will lead to a quality increase when using combing. But, this might just be an edge case, with empty infill.

fvrmr, 2021-03-01:
Hi @CCS86 could you share your project file? 
So I can see what is happening. Cause I can't see it with your file, because the settings are missing.

CCS86, 2021-03-01:
> Hi @CCS86 could you share your project file?
> So I can see what is happening. Cause I can't see it with your file, because the settings are missing.

It's at the bottom of the original post.

fvrmr, 2021-03-01:
> It's at the bottom of the original post.

Yeah, I downloaded that file. But it was just a 3mf file, because dragging it into Cura this didn't pop up and it should do that with a project file. 
[image]

Actual result:
[image]

So in this I use my printer and settings. And with a project I have your printer and settings, so it is easier to reproduce.

CCS86, 2021-03-01:
[Coast-bug2.3mf.txt](https://github.com/Ultimaker/Cura/files/6063792/Coast-bug2.3mf.txt)

fvrmr, 2021-03-02:
Thanks, I understand and see what is happening. 
It doesn't matter how many wall lines you have it will always move to the 3rd wall/ 2d inner wall.

I will discuss it with the team. Keep you posted.

Ghostkeeper, 2021-03-05:
To summarise what we've discussed with the Cura team:

* To Cura, 0% infill is still the inside of the model. It doesn't know that for your intent the inside of your walls also need to be pretty.
* It seems to move the nozzle away from the outer wall to prepare for a retraction, perhaps for the layer change. This is allowed to go towards the inside, after all, retracting while on the outer wall is bad. You're better off moving inside. So this would be expected behaviour and not a bug.
* We realise the desire that the inside of the model might need to be made pretty as well, but that's not considered to be worth the development time or the added complexity to the interface or settings.

Loading the project file into Cura myself though I see that it's slightly different. What's happening is that it's travelling to the starting position of the next layer. This travel move is a combing move and happens to be almost to the same spot. Because the combing strategy tries to avoid being on the outer wall at all cost, and being on the first inner wall secondarily, it moves away from the 2 outermost walls here first, then goes as close as it can to its destination, then travels back onto the outer wall to continue printing there.

I think in a perfect world it would see that the path to the destination is shorter than the path away from the outer wall and then make a shortcut to the destination without combing, as the lesser of two evils.

But I also agree with the Cura team that this case is not going to need the priority from us now due to the low impact, so we'll still have to defer it. We can reconsider if it turns out to be very easy to fix or if it turns out to have a bigger impact.

closed 2023-06-24 by github-actions[bot]