Ultimaker/Cura #9341: Slicing errors in version 4.8

jornada812, 2021-02-27:
version 4.6.2
[image]

version 4.8 with imported profile from 4.6
[image]

version 4.8 
[image]
default profile 0.1 

stlmodel - https://yadi.sk/d/kEV_49wS19pz-Q
profile from version 4.6 - https://yadi.sk/d/T2zzspP_aYvb0w

GregValiant, 2021-02-27:
Can you post the STL here?  I don't install software just to download a file.

jornada812, 2021-02-27:
What software do you not want to install? These files are available for saving by link. any whim of yours

GregValiant, 2021-02-27:
That was my bad.  There was a popup over the download button and all I noticed was "Install yanex".

jornada812, 2021-02-27:
I close the task. This defect occurs when the function of "Coasting" is active, which for some reason is enabled by default in the experimental section

closed 2021-02-27 by jornada812

GregValiant, 2021-02-27:
Did you have Coasting enabled?  If you have the preview set to show extrusions there is no extruding during a coast so there are no lines there.  In the profile though, coasting was off.  When I turned it on, the preview looked like your screen shots.

jornada812, 2021-02-27:
in profile 4.6 the coasting was turned off, I never used it. But after importing to 4.8 for some reason it's included. And in all profiles by default in version 4.8 it is also enabled for some reason.

Ghostkeeper, 2021-03-05:
Since you didn't fill in the template, I can't check if people changed the default profiles for your configuration.

It could also be that you loaded in a project file or a profile from someone else that enabled the setting?

jornada812, 2021-03-05:
Well, it doesn't matter anymore, the main thing is that the reason is found