Ultimaker/Cura #9342: Nozzle diameter change is not saved in the printer profile

jornada812, 2021-02-27:
[image]

and the default config file (in my case predator_extruder_0+%232_settings.inst.cfg) does not contain a parameter machine_nozzle_size,  i have to add it manually

fvrmr, 2021-03-01:
Hi @jornada812 thank you. I am not familiar with Anycubic printers. When I add a Anycubic printer in Cura the nozzle size is 0.4 mm. We don't maintain the config files for these printers, so if it needs to be changed by a pull request.

Please let me know if this is the case, if not could you provide me your config file and more information.
Thank you in advance.

jornada812, 2021-03-01:
I didn't understand your answer. In any printer, the user can change the
nozzle. Put 0.3, 0.6 or any other value. The interface does not allow you
to remember the new nozzle in any printer model, even branded, even custom.
I suggest you fix it or describe the algorithm, as I can do it in the
interface. This error moves from version to version. Sometimes they write
that it was fixed, but it comes back again

fieldOfView, 2021-03-01:
Try entering the value with a comma instead of a dot.
This has been fixed for Cura 4.9

jornada812, 2021-03-01:
> Try entering the value with a comma instead of a dot.
> This has been fixed for Cura 4.9

It works. thank

closed 2021-03-01 by jornada812