Ultimaker/Cura #9343: wierd supports behavior

RafaeleValanten, 2021-02-27:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

jellespijker, 2021-02-27:
You have peeked my curiosity. I would love to see that weird support behaviour. Does it dance, time travel or maybe it turns into cheese?

Please fill in the issue template and provide log files and project files.

jellespijker, 2021-02-28:
Thank you for filling in the issue report. Could you also add the project file.

RafaeleValanten, 2021-02-28:
> Thank you for filling in the issue report. Could you also add the project file.

> Thank you for filling in the issue report. Could you also add the project file.

may i ask what else should i paste? the status sais that "needs info"

jellespijker, 2021-03-01:
My apology, I overlooked the project file. I think this should be enough information for us to investigate further.

FYI @fvrmr

fvrmr, 2021-03-01:
Thank you @RafaeleValanten, is this behaviour failing your print? 
I am able to reproduce this, so I will discuss this with the team. Keep you posted.

RafaeleValanten, 2021-03-01:
> Thank you @RafaeleValanten, is this behaviour failing your print?
> I am able to reproduce this, so I will discuss this with the team. Keep you posted.

Actually not.. It is only makes blobs on the outer wall in places where this supports line approaching the outer wall , cz i set till 5 mm no retraction.

fvrmr, 2021-03-02:
Hi @RafaeleValanten I have discussed it with the team. And we have to defer this issue, because conical support is in experimental mode it is not a priority for us right now. Hope you understand.

closed 2023-06-13 by github-actions[bot]