Ultimaker/Cura #9344: Coasting Inconsistent with 1 Wall

CCS86, 2021-02-27:
4.8 / Win10 x64

When trying to make a hollow cylinder with one wall, coasting is only triggered on some of the layers:

[image]

Add a second wall and it triggers on every layer:

[image]

[Coast-bug.3mf.txt](https://github.com/Ultimaker/Cura/files/6055282/Coast-bug.3mf.txt)

fvrmr, 2021-03-01:
Hi @CCS86 thank you for your bug report. 
Did you try to enable to the surface mode? If that doesn't work for you, could you please share a project file (file - save project) then I can try to reproduce your problem.

CCS86, 2021-03-01:
I have not tried surface mode, but the project is at the bottom of the original post.

fvrmr, 2021-03-01:
Hi @CCS86 can you let me know if that works for you. If not share a project file, because that file is also just a 3mf file.

CCS86, 2021-03-01:
Sorry about that. I must have chosen export. See if this works:

[UMO_Circle2.3mf.txt](https://github.com/Ultimaker/Cura/files/6063762/UMO_Circle2.3mf.txt)

fvrmr, 2021-03-02:
Thanks, I am also able to reproduce this. 
Will also discuss this with the team. Keep you posted.

Ghostkeeper, 2021-03-05:
We discussed this with the Cura team and found it worthy to give this some priority. We couldn't reproduce this with different models but there's a chance that it happens with other models as well. While the coasting feature isn't used a whole lot (at least not on purpose) we also don't think that this is all too difficult to fix in general. @konskarm don't forget to create a ticket in Jira!

closed 2024-12-08 by github-actions[bot]