Ultimaker/Cura #9345: Parabolic Infill and Top/Bottom Pattern

bradsoto, 2021-02-28:
Typically the first limit a robot sees is acceleration (stepper motor torque plus bed mass for example), followed by velocity (extrusion slip). To minimize print time I would like to align the infill direction in the fastest accelerating orientation (X for cartesians, X+Y for coreXY). To keep this explanation simple I'll tend to focus on cartesian bots. If we align the infill path to X the hotend will spend the most time at the top velocity, especially if the longest orientation of the part is also aligned with X. To maximize the amount of material applied during the X pass there should be oscillating acceleration in the Y direction. The maximum oscillating Y acceleration will occur only at maximum X velocity, and so an optimal "infill line distance" and "infill wavelength" would be dependent on the maximum Y acceleration and maximum print speed of the user's particular printer type/material/extruder. It is assumed that the user would find the limits of their setup themselves and calculate the optimal sine wave amplitude and wavelength, creating the best-case scenario of reaching maximum Y acceleration at maximum X velocity. For both infill and top/bottom walls there needs to be another setting "wavelength", and a setting similar to "infill line distance" for top/bottom pattern. In both cases once reaching the end of a line we would should continue extruding like "zigzag" and "gyroid". Speaking of gyroid it could also benefit from the extra "wavelength" setting in cases were the infill orientation does not matter like delta bots. CoreXY could also benefit from an the additional option to rotate the pattern 90 degrees on alternating top/bottom walls.

For top/bottom walls these sine wave shapes would be stacked on top of each other like pringle chips, but for infill we would flip every second path. The infill paths will meet each other on every wavelength, so the amplitude increases as "infill line distance" increases. If a user changes "infill density", Cura adjusts "infill line distance" and "wavelength" together. That's basically it, I hope someone else will consider this a useful feature and give it an upthumb.

fvrmr, 2021-03-01:
Hi @bradsoto thank you for your feature request. 
Could you provide us a drawing of how this should work, so we can visualize your request.

bradsoto, 2021-03-01:
[image]

At first it looks like there is an issue with overlaps and gaps for top/bottom walls, but if we keep the extrusion rate the same throughout it should even out. Where there are overlaps the hotend will be moving faster, and where there are gaps the hotend is moving slower. In this drawing the line thickness is constant, but in reality it will be thickest at the peaks and valleys, while thinnest in the middle.

fvrmr, 2021-03-02:
Thanks, I am not sure what you mean with top/bottom walls. Do you mean top/bottom pattern instead?

bradsoto, 2021-03-02:
Oh yes I do.

fvrmr, 2021-03-02:
Thanks, I will discuss it with the team. Keep you posted.

fvrmr, 2021-03-05:
Hi @bradsoto I have a few more questions.
Why specific do you want this infill and patterns and for what kind of models you want this wall pattern? Is it only for coreXY printers, or is it also for other printers? And why do you believe that this will work. 
Your infill looks a bit like honeycombs is that something that will work for you? 
Hope you can answer these questions for me.

bradsoto, 2021-03-06:
> Why specific do you want this infill and patterns and for what kind of models you want this wall pattern?

It will print faster.
> Is it only for coreXY printers, or is it also for other printers?

Any printer will see a speedup. Less so for deltas more so for cartesians.
> Your infill looks a bit like honeycombs is that something that will work for you?

Not really, honeycomb is slow.

Ghostkeeper, 2021-03-08:
Finally taking a close look at this. This pattern has some of the same merits as a proposed honeycomb pattern, but will probably turn out to be a bit less stiff since there are no straight edges to lean on. Instead it should feature less acceleration. Yet in toughness I don't see how this could beat Gyroid. I do have some ohter concerns, especially considering the intended use case of this as printing fast.

The aim of this, you state, is to extrude material faster. If going horizontal at maximum speed, you can increase the vertical speed slightly by going up and down, resulting in faster extrusion. However there is an even better way to extrude fast: Also maximise the vertical speed. So just print a diagonal straight line. This is what Lines and Grid infill do. By default they go diagonal, so that the printer can accelerate faster and (if the line is long enough and the extrusion train can keep up) reach greater speeds.

I don't think this pattern would be appropriate for top/bottom. For one, you usually want to extrude those slower. But also because they have overlaps. Indeed, in theory the material would flow over to the side and fill exactly the available space. But in practice, as you go over previously printed lines you will increase the backpressure going into the nozzle and you will briefly block the flow. This then takes a while to get flowing again. As a result, you'll see many gaps in your pattern. This is why Triangles is a terrible pattern to print at high infill densities (>50%). Yes, the amount of material would fill exactly the required amount of space, but due to this backpressure effect you're going to get heavy underextrusion throughout.
So no skin. But for infill this pattern should be fine, at least below 50% or so.

But basically, I don't see how this extrudes faster than Grid, ZigZag or Lines. And I don't see this being better than Gyroid in toughness, or better than Tri-hexagon in stiffness. Of course, nothing like an actual print test though.

closed 2021-03-10 by bradsoto