Ultimaker/Cura #9348: Extra prime after combing

RafaeleValanten, 2021-02-28:
**Is your feature request related to a problem? Please describe.**
Underextrusion after combing. 
 So i use to print big lithophanes. Combing is a must here cz i save a few hours with it while filing gaps between walls without retractions. (Optimize wall printing order is on)
Underextrusion appears on a start of the new layer. 
 Settings   : 
- Combing (Distance 40 mm)
- Fill gaps between walls
- Optimize wall printing order
- retract before outer wall
- extra prime 0.1 ( if i set more it will fix the problem on the start line of the new layer but will make blobs in places where no need in extra prime.
Well, how does it work:
1) cura makes all walls of the cylindrical 35 cm lithophane 
2) then starts to filing gaps and do this along all circle working with combing and without any retractions. So nozzle loses a lot of pressure/material while traveling between gaps.
3) finally after all circle is done it makes one retraction and moves to start line of the new layer.
4) Starts to print that new layer line with underextrusion.

**Describe the solution you'd like**
Extra prime amount after every combing travel.

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

**Additional context**
(Add any other context or screenshots about the feature request here.)

fvrmr, 2021-03-01:
Hi @RafaeleValanten thank you for your feature request. 
I will discuss this with the team. Keep you posted!

CCS86, 2021-03-01:
I would add that basing the amount of extra prime on "max volume lost" and "combing time" factors would give better control, compared to a fixed prime volume.

I would guess that the oozed volumetric flow curve probably looks like exponential decay, where ooze flow starts high and reduces quickly.

Whether the comb move is over the new layer, or the last layer probably plays a big part too. And whether the new layer is sparse or solid infill.

RafaeleValanten, 2021-03-01:
> I would add that basing the amount of extra prime on "max volume lost" and "combing time" factors would give better control, compared to a fixed prime volume.
>
> I would guess that the oozed [...]

that's a great idea!

fvrmr, 2021-03-02:
Hi @RafaeleValanten I also discussed this issue with the team. We all think this has value but right now our priorities are at our new release LibArachne. So that's why I have to defer this issue. Hope you understand.

RafaeleValanten, 2021-03-02:
> Hi @RafaeleValanten I also discussed this issue with the team. We all think this has value but right now our priorities are at our new release LibArachne. So that's why I have to defer this issue. Hope you understand.

ok, thank you

cameronbergh, 2021-04-17:
i really think a bunch of the under extrusion problems would be fixed if there was a extra prime after travel movements, and rather than setting a fixed amount, make it a function of the distance of the last travel move?

Triangulix, 2022-01-20:
I have the same problem. I'm printing a cylinder with 1.2mm thick walls so that it is exactly 3 lines wide.
Cura prints the inner wall, then the outer wall on the inside of the cylinder. Then it inserts a stupid travel move all around the cylinder before it starts printing the outer wall on the outside. The nonsense travel move pulls lots of material out of the nozzel and the start of the outside wall gets almost no material on the first 20mm.

"Retraction Extra Prima Amount" has no effect since there is no retraction.
"Retract Before Outer Wall" has no effect. No retraction is happening. I'd say the function is broken.
"Retract at Layer Change" doesn't help either since the layer starts with the inside wall.
"Max Comb Distance With No Retract" has no effect. The travel around the object is probably not seen as combing move.
"Outer Before Inner Walls" helps with this problem but ruins another part of the print.

Using 1.6mm as wall thickness, resulting 4 wall lines, eliminates the useless travel move but increases weight of the object which I try to keep low.

Ghostkeeper, 2022-01-27:
> I have the same problem. I'm printing a cylinder with 1.2mm thick walls so that it is exactly 3 lines wide.
Cura prints the inner wall, then the outer wall on the inside of the cylinder. Then it inserts a stupid travel move all around the cylinder before it starts printing the outer wall on the outside. The nonsense travel move pulls lots of material out of the nozzel and the start of the outside wall gets almost no material on the first 20mm.

Should be noted that Cura 5.0 doesn't create that 4th wall any more at all in this situation.

Triangulix, 2022-01-29:
> Should be noted that Cura 5.0 doesn't create that 4th wall any more at all in this situation.
Which 4th wall do you mean? With 1.6mm wall thickness 4 wall lines should be correct.

Ghostkeeper, 2022-02-14:
> Which 4th wall do you mean? With 1.6mm wall thickness 4 wall lines should be correct.

Not if the part is 1.2mm thick, as you mention.

Triangulix, 2022-02-14:
> > Which 4th wall do you mean? With 1.6mm wall thickness 4 wall lines should be correct.
> 
> Not if the part is 1.2mm thick, as you mention.

Do you mean that what I understand as a travel move is actually an additional wall that is printed on the same position as an existing wall and without extrusion?

Ghostkeeper, 2022-02-26:
> Do you mean that what I understand as a travel move is actually an additional wall that is printed on the same position as an existing wall and without extrusion?

I have no way of knowing whether it is since you didn't give a project file to show your settings, but it is indeed likely in Cura 4.x if you were using the defaults for most popular printer profiles. When presented with a part slightly wider than 3 line widths, Cura 4 creates 2 contours, one inset by 0.5 line widths and another inset by 1.5 line widths. These are the centrelines of the two contours in the part. It will print those contours. They are walls. Walls are loops, so they always close completely. You get 2 walls, so 2 loops around your long-and-thin part. That may look like 4 walls because each loop is going back and forth. But they are 2 looping walls according to Cura.

However there is also the feature "Compensate Wall Overlaps". If enabled, it will see that the innermost loop (almost) completely overlaps with itself, and reduce the flow rate of one half of the loop to prevent overextrusion. What you get then is that one half of the loop is printing normally, and the second half is printed with like 0.01% flow, effectively turning it into a travel move. But it's still going to print that wall, even though it has 0.01% flow.

There is also the Minimum Wall Flow setting, which prevents printing parts of the wall beneath a certain flow rate. So you could set that to 30% or something, and extremely thin parts of the wall won't get printed then, effectively leaving you with the 3 lines you intend. One of those 3 lines is actually 0.01% too thin then, but you won't notice that in the print.

In Cura 5.0 this all won't be necessary since it will adjust the line width to make it fit the shape with exactly 3 walls.

Triangulix, 2022-02-26:
> I have no way of knowing whether it is since you didn't give a project file to show your settings, but it is indeed likely in Cura 4.x if you were using the defaults for most popular printer [...]

Thanks for the clarifications but I believe you are talking about another issue which I also experience but usually solve by setting the line width to 0.39 instead of 0.4.

I have attached 2 projects. Please watch the nozzle's path on layer 40 of each of the projects.
[Testzylinder.zip](https://github.com/Ultimaker/Cura/files/8146611/Testzylinder.zip)

Project "Testzylinder, 3 Bahnen, medium" shows the issue you probably mean. This is resolved by setting line width slightly smaller.
Project "Testzylinder, 3 Bahnen" shows the issue I meant. After printing the innermost wall the nozzle goes around the object before starting the outermost wall. This leads to the nozzle being empty when the outermost wall starts, leading to underextrusion on the fiirst 15mm of this wall.

Triangulix, 2022-03-11:
I can now confirm that neither disabling "Compensate Wall Overlaps" nor setting "Minimum Wall Flow" to 30% changes anything about the reported behavior. Cura in this version simply cannot handle round shapes with exactly 3 wall lines correctly.

Ghostkeeper, 2022-03-14:
No, the "Testzylinder, 3 Bahnen" shows exactly what I meant. One of the lines is almost completely overlapping the other. It's not a travel move (doesn't show up in layer view as a travel move when those are made visible), but it might as well be with that much overlap compensation.

Here is a project file that shows that cylinder printing with 3 lines and no roundabout travel moves: [3-wall-cylinder.zip](https://github.com/Ultimaker/Cura/files/8241156/3-wall-cylinder.zip)

"Testzylinder, 3 Bahnen, medium" shows this issue: https://github.com/Ultimaker/Cura/issues/7687

The issue we were both talking about is this issue: https://github.com/Ultimaker/Cura/issues/7656

Both have been fixed in 5.0. The original issue in this thread is about an extra prime after combing though, so let's get back on topic.

Triangulix, 2023-06-29:
Since most user input has been marked as off-topic I'm not going to waste another minute on helping to improve Cura.