Ultimaker/Cura #9349: 4.8.0 unable to drag and drop stl files

charj, 2021-02-28:
**Application version**
4.8.0 Windows 10

**Platform**
Windows 10 / i7 8700k / RTX 3080 

**Printer**
Ender 3 Pro - though not relevent here

**Reproduction steps**
1. Download a stl file (tried many)
2. Open cura
3. Drag file into cura

**Actual results**
Dragging files into cura has no result

**Expected results**
I expect files to be added to the build area upon drag

fvrmr, 2021-03-01:
Hi @charj could you please add your log file and maybe a screenshot or video, so I can see what exactly is happening. 
You can find your log file here: `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

charj, 2021-03-01:
Please see video here:
https://www.youtube.com/watch?v=V832fzho7c0&feature=youtu.be

charj, 2021-03-01:
Here's my log

`2021-03-01 13:21:00,352 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [710]: Slicing took 1.3041636943817139 seconds
2021-03-01 13:21:00,352 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [711]: Number of models per buildplate: {0: 1}
2021-03-01 13:21:00,352 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [725]: See if there is more to slice...
2021-03-01 13:21:00,356 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [196]: Socket state changed to Closing
2021-03-01 13:21:00,359 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [220]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
2021-03-01 13:21:00,362 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [198]: Socket state changed to Closed
2021-03-01 13:21:00,372 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [898]: Backend quit with return code 0. Resetting process and socket.
2021-03-01 13:21:06,723 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [48]: Scheduled the next on-app-exit callback [<bound method USBPrinterOutputDevice._checkActivePrintingUponAppExit of <USBPrinting.USBPrinterOutputDevice.USBPrinterOutputDevice object at 0x0000017F1D904678>>]
2021-03-01 13:21:06,727 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [51]: No more on-app-exit callbacks to process. Tell the app to exit.
2021-03-01 13:21:06,730 - INFO - [MainThread] cura.CuraApplication.closeApplication [603]: Close application
2021-03-01 13:21:06,745 - DEBUG - [MainThread] UM.Qt.QtApplication.windowClosed [470]: Shutting down cura
2021-03-01 13:21:06,800 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [315]: Attempting to kill the engine process
2021-03-01 13:21:11,261 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [48]: Scheduled the next on-app-exit callback [<bound method USBPrinterOutputDevice._checkActivePrintingUponAppExit of <USBPrinting.USBPrinterOutputDevice.USBPrinterOutputDevice object at 0x0000023F20739798>>]
2021-03-01 13:21:11,266 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [51]: No more on-app-exit callbacks to process. Tell the app to exit.
2021-03-01 13:21:11,268 - INFO - [MainThread] cura.CuraApplication.closeApplication [603]: Close application
2021-03-01 13:21:11,277 - DEBUG - [MainThread] UM.Qt.QtApplication.windowClosed [470]: Shutting down cura
2021-03-01 13:21:11,366 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [315]: Attempting to kill the engine process
2021-03-01 13:21:11,366 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [321]: Killing engine process
2021-03-01 13:21:11,371 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [324]: Engine process is killed. Received return code 1
2021-03-01 13:21:28,388 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-03-01 13:27:26,591 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [130]: Writing to [K:/Ender 3 Parts/Drawers/CE3PRO_CE3_Left_Lower_Drawer_124mm.gcode]...
2021-03-01 13:27:26,595 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._parseJWT [100]: There was no auth data or access token
2021-03-01 13:27:26,599 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [148]: Writing to Local File K:/Ender 3 Parts/Drawers/CE3PRO_CE3_Left_Lower_Drawer_124mm.gcode in text mode
2021-03-01 13:27:27,065 - DEBUG - [Thread-7] UM.FileHandler.WriteFileJob.run [80]: Writing file took 0.45600152015686035 seconds
2021-03-01 13:27:27,108 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [284]: SliceInfo sent successfully
2021-03-01 13:27:57,073 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.8.0/resources/qml/RoundedRectangle.qml:68: TypeError: Cannot read property of null`

Ghostkeeper, 2021-03-02:
So far I'm unable to reproduce this issue. Dragging and dropping files onto the Cura screen works fine for me.

That is two snippets of a log file, showing a place where a slice was completed and you closed down Cura. And then a place 7 minutes later where you saved g-code to disk. Presumably the dragging/dropping of a model happened before the first snippet and/or in between the two, but that was cut out of the log.

Could you share a snippet of the log file during the entire run of Cura when you dragged a model onto the build plate?

The video link is private and not accessible, it seems.

fieldOfView, 2021-03-02:
Are you running Cura "as Administrator"? If so that is the problem. An application that is run "as Administrator" can not receive a drop event from an application that is not run "as Administrator", like Explorer.

It should not be necessary to run Cura "as Administrator". If it is, then that should be the issue to report.

charj, 2021-03-02:
Ok, this is a little embarrassing.

I was dragging the files from a zip file directly. If I extract it to a folder and drag those files it works as expected.
This is due to me recently reinstalling Windows and not being used to the default behaviour of extracing zip folders.

I'm not totally sure why this wouldn't work although I guess it could potentially be supported?

closed 2021-03-02 by charj

Ghostkeeper, 2021-03-08:
I think the zipped-folder archiver is specifically looking for a file location to unzip the file to. It can't unzip the file in memory and directly pipe it to the application or anything.