Ultimaker/Cura #9350: Make Overhangs Printable not being affected by cutting mesh.

echan42, 2021-03-01:
**Application version**
4.8.0

**Platform**
Windows 7 x64 

**Printer**
Ender 3

**Reproduction steps**
I'm having trouble using "make overhang printable" on specific parts of my model.
I do the following:

1.     Load model, tick "make overhang printable".

2.     Add cutting mesh to part I don't want "make overhang printable" to be applied, un-tick "make overhang printable" on the      cutting mesh options.
 
3.     Slice and observe that the whole model has overhangs printable rendering the lower pegs useless.

**Screenshot(s)**
https://preview.redd.it/k5m1sn491ek61.png?width=1419&format=png&auto=webp&s=00c81a7a9ffd268a87e8aba008f7d93ef6ed2194

https://preview.redd.it/55xfy7at1ek61.png?width=1634&format=png&auto=webp&s=f06f6bf93af3aa8a278147a8c303e0707bff487b

**Actual results**
Printable overhangs are applied to the whole model, rendering the lower pegs unusable.

**Expected results**
Lower pegs should be have regular support, rest of the model should have printable overhangs.

**Project file**
https://wetransfer.com/downloads/f7d1b4f55c491839834045a67c4c459220210301143722/645e7b

fvrmr, 2021-03-01:
Hi @echan42 thank you for your report. 
I will discuss this with the team. Keep you posted!

fvrmr, 2021-03-02:
Hi @echan42 I have discussed it with the team. But we don't get really your issue.
How do you want this to be printed? And why do you want regular under the lower pegs?

echan42, 2021-03-03:
Hey, please bear with me if I don't explain myself properly.
Is it ok if I link a drawing? 
[image]

fvrmr, 2021-03-04:
Thanks for you explanation, a drawing is always good to explain things. 
But I am curious why do you want to print it like this?

Ghostkeeper, 2021-03-09:
Perhaps because the piece needs to fit in one place, but the details on the side are not important for that fit. I think the intention is clear enough.

Ghostkeeper, 2021-03-09:
This happens because Make Overhang Printable is applied as one of the first steps in slicing, even before the cutting meshes are separated. To fix this, we'd need to change the order around of some things there.

fvrmr, 2022-03-21:
This issues has been discussed again with the team and we decided to defer this issue. It has no priority for us to implement, also Cura is not a the right program to make adjustments to a 3D model, if you want changes it is best to do this with a CAD program. And it has also to do with what @Ghostkeeper described here above, anyone is free to implement this issue and create a pull request.

JTeasdaleMEC, 2025-02-04:
Hi,

I am using Cura 5.9.0 and this still isn't resolved. I saw issue #11670 which is exactly what I am experiencing, but have been closed as being a duplicate of current issue.

I understand that Cura isn't a proper modelling program, but having `Make Overhang Printable` is a nice and handy feature for performing minor geometry adjustments on the fly and avoid wasting time and material printing unnecessary supports. For instance, I am working on a model which most part of it have been thought to be printable, but two features requires supports to be correctly printed. Being able two use support block modifiers to locally apply `Make Overhang Printable` would perfectly suit this situation.

Parts to be printed with circled in blue the part to which I want to locally apply `Make Overhang Printable` :
[image]

Support block modifier used : 
[image]

Desired result : 
[image]

Actual result : 
[image]

And globally applying `Make Overhang Printable` and trying to locally disable it using a support block modifier isn't working either. `Make Overhang Printable` isn't affected by support block modifiers.

Hopefully the screenshots are helping to make the explanation clear. 🤞 

Is this something that can be corrected please?
Best Regards