Ultimaker/Cura #9351: Coordinate Walls with Infill

bradsoto, 2021-03-01:
Another feature request to reduce print time. "Coordinate Walls with Infill" should do these things:

### Alternate Between Inner and Outer Walls
An inner wall first layer is first followed by an outer wall first layer. The end of the inner wall first layer is the beginning of the outer wall first layer.
### Coordinate Infill and Double Infill Layers 
The start of the infill should be the end of the outer wall first layer. This would mean changing "Infill XY offset" for simple infills, or changing the start point of the inner wall first layer. If infill layer thickness equals the wall layer thickness then we should do 2 infill layers, following the first infill layer path in reverse. If infill layer thickness is 2x wall layer thickness we should only do 1 infill layer. In both cases the end of the infill layer(s) is the start of a new inner wall first layer.

Repeat the process for the entire shape. For simple shapes and infills this will reduce print time by eliminating/minimizing non-extruding moves. For complex infills this can reduce the Z seam by moving the start of a new inner wall first layer around a bit. The same features should be applied to top/bottoms walls.

The first layer should start with the bottom wall. The end of the bottom wall is the beginning of an inner wall first layer. For slices with multiple islands we should only move between islands after 2 wall layers and 2 infill layers. This will also reduce total string volume by half.

Ghostkeeper, 2021-03-02:
Currently Cura has a setting "Infill Before Walls" which is a trade-off. One could imagine this as a third option to the currently boolean value of that setting.

When FFF printing, you typically want your line to extend slightly horizontally beyond the normal line width. If that line is then placed next to another line, they'll fuse together. If this doesn't happen, the rounded shape of a layer causes the line to not touch the next line at all, basically ruining the strength of the part. With a good print (not underextruding), there is effectively a slight overlap between adjacent lines, which makes the two lines touch with some surface area, which makes them join together and gain strength.

When printing the infill first, due to this overlap, the walls will be pushed outward slightly by the already solidified infill lines. Conversely, when printing the walls first, the infill will be pushed inward slightly while the walls stay in place, since the walls have already solidified by the time the infill gets printed.

As a result, if this feature is enabled, you'll get a rougher surface because the wall will be pushed outwards on alternating layers.

I also don't think the print time savings will be very great. On a typical print, the total travel time is 2-5% of the total printing time. If we are optimistic and say that this can remove 30% of the travel time, you'd be saving ~0.5% to 2% of the printing time. On a 4 hour Benchy print that's between 1 and 5 minutes.
The greater benefit would be to reduce flow changes. The flow change between infill and walls is the most problematic one of the whole print, especially when going from infill to the outer wall (when printing Outer Before Inner Walls) because the flow change will be visible on the outside and impact dimensional accuracy. Reducing the number of flow changes would help reduce the amount of blobs on the surface, for which this would be a useful feature.

Finally, I think the benefit will be destroyed the moment there is any skin or support in the layer, you're using Infill Layer Thickness, multiple parts or dual extrusion. For some parts of a simple model it would work. But for many cases not.

So before we consider implementing anything like this, I feel like these issues need to be addressed in some way. See if we can find a way to work with them or around them.

bradsoto, 2021-03-10:
Yup I think decreases flow changes overall would be better strategy than this.

closed 2021-03-10 by bradsoto