Ultimaker/Cura #9353: 2. to be able to see graphical print progress

VladekRo, 2021-03-01:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

**Additional context**
(Add any other context or screenshots about the feature request here.)

closed 2021-03-01 by VladekRo

Ghostkeeper, 2021-03-09:
There's a progress bar, if you're connected to the printer, right?

VladekRo, 2021-03-09:
nop, picture is static

Growing Old is Compulsory. Growing Up is Optional.

Democracy: Where two idiots outvote a genius???

fieldOfView, 2021-03-10:
I could take a guess that what @VladekRo is requesting is a simulation-view-like display of the layers as they are printing over USB, and that the progress-bar is not sufficient for what he'ld like to see. I'm pretty sure this request is a duplicate of a previous request, but I'm too lazy to search for prior requests.

VladekRo, 2021-03-10:
Aldo
Just look what Repetier shows during printing, thank you.
Best Regards
Vlad

Growing Old is Compulsory. Growing Up is Optional.

Democracy: Where two idiots outvote a genius???

Ghostkeeper, 2021-03-23:
Duplicate of https://github.com/Ultimaker/Cura/issues/6020.