Ultimaker/Cura #9354: Make Cura Aware of Hotend Geometry

bradsoto, 2021-03-01:
Once feature #9351 is complete, we become very motivated to build up a single island as high as we can without taking a break to build up the other islands. This feature is similar to one-at-a-time print sequence, although much more difficult due to islands being so close to another on the same object. Another similar feature is used with non-planar slicing, although "maximum angle" is only a rough approximation of hotend geometry and only allows minimal Z freedom. Ideally we should allow the user to import a 3D file of their hotend geometry and calculate collision avoidance using the real-world geometry. The slicer should spend the maximum time building a single island as possible to minimize non-extruder moves. Also we should have a "maximum tower time" setting if we have a limited time before layer cooling starts impacting print quality for some material.

Ghostkeeper, 2021-03-02:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/6642. We call the feature "Towering" and it has been attempted before with limited success in the CuraEngine code base. It failed on the cooling back then.

There are a couple of limitations we have to think about here. One that you already mentioned is that cooling and minimum layer time become much more of a problem. But the most obvious problem is inverse kinematics. Allowing a free-form 3D model to function as the print head shape is problematic because it basically requires the slicer to predict where the print head will be throughout its motions and create optimal paths based on that, which is very computationally expensive. The case where the print head is fixed in its orientation is relatively simple, but still requires a 3D Minkowski sum of arbitrary 3D models at the very least (the sum of the shape of the print head with the shape of a supposed path for the print head).

I've done Minkowski sums of 3D volumes (CSG) in OpenSCAD that take a full day to compute even if the model isn't all that detailed. And Cura doesn't even read CSG volumes, just surface meshes, for which there **is no reliable algorithm as of this writing** to do Minkowski sum unless the surface is completely manifold, which the user can't guarantee. So I don't think that is really an option.

A much more feasible approach would be to assume the shape of the nozzle to be a cylinder with radius `r` and the nozzle protruding below the print head with a height `h`. You could then group all parts of a volume together that are closer than `r` horizontally, within a space of layers spanning `h` vertically. Print those at the same time and then move to the next group horizontally.

The feature has been deferred though. It is not considered to be a priority due to the trade-off in complexity vs. how much gain you'll really get from it. It'd reduce stringing a bit at the cost of sagging and hitting minimum layer time more often. The travel moves are not really going to bring a significant benefit except with very extreme models.

closed 2021-03-02 by Ghostkeeper

bradsoto, 2021-03-02:
Could we slice the hotend geometry and temporarily add those extra slices for collision detection?

Ghostkeeper, 2021-03-09:
Sort of, yeah. The 2D Minkowski sum is at least a solved problem in Computer Science. You'd need to union the collision area for all relevant layers though. Even for a 5cm tall print, that's 500 Minkowski sums and 500 unions, just to calculate the collision area for a single layer! You can't cache them either, since on the higher layer the Minkowski sums will be different because the print head is higher then.