Ultimaker/Cura #9355: Please include layer by layer preview in "monitor" while printing via USB

RaminShalom, 2021-03-01:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

**Additional context**
(Add any other context or screenshots about the feature request here.)

RaminShalom, 2021-03-01:
While printing via USB & in the "Monitor" window, please include the layer by layer preview

fvrmr, 2021-03-02:
Hi @RaminShalom please fill in the form. So we can understand what you exactly want.
Thank you in advance.

RaminShalom, 2021-03-03:
I would like to see a layer by layer graphical representation of the object while being printed via USB

fvrmr, 2021-03-05:
Why would you want this? Is this related to a problem? See the form below for the questions we need before discussing this with the team. 

> Is your feature request related to a problem? Please describe.
> (A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])
>
> Describe the solution you'd like
> (A [...]

Ghostkeeper, 2021-03-09:
I think he means something like this? https://github.com/Ultimaker/Cura/issues/6020

RaminShalom, 2021-03-09:
Absolutely correct,

 

The gcode can be translated to the visual representation as the print progresses.

fvrmr, 2021-03-09:
As you can see in that thread we've decided to defer this issue. Because we don't support USB printing on our printers anymore. 
A pull request is always welcome off course. 
Hope you understand.

closed 2021-03-09 by fvrmr