Ultimaker/Cura #9356: more fence lines than installed

sibmaker, 2021-03-03:
**Application version**
(4.8.0)

**Platform**
(windows 10. GeForce 2070 Super)

**Printer**
(Flying Bear Ghost 5)

**Reproduction steps**
1. (0.4 mm nozzle)
2. (Selected 3 wall lines)
3. (Sliced the model)

**Screenshot(s)**
(
Before
[image]
After (prints this line again)
[image]
) 

**Actual results**
(Walls with an even number of lines print well, with an odd middle line printed 2 times. For example, on the part below, the 1.6 mm wall printed well, and 2 is already bad.
[image]
)

**Expected results**
(The middle wall must be printed 1 time, otherwise the quality will deteriorate)

**Project file**
(
[Example.zip](https://github.com/Ultimaker/Cura/files/6073313/Example.zip)
)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(This problem has been present in CURA for a long time and spoils all the quality of this wonderful platform. I hope for the answer from the developers, maybe my settings are not correct. And hope for quick prevention measures.)

fvrmr, 2021-03-04:
Hi @sibmaker thank you for your report. 
I am able to reproduce this in 4.8 as well, but I sliced it also in our Arachne release. And there is doesn't happen. 

[image]

You can read more about our Arachne release [here.](https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_alpha)

sibmaker, 2021-03-05:
Hello!
Try it in 4.8. Since I have it happening there. And if you are doing well in 4.8, tell me why is it so? What could I put to have such an effect? Since the problem is relevant and not only for me and I plan to make a video on this topic on YouTube

  
>Четверг, 4 марта 2021, 21:22 +07:00 от Fenne <notifications@github.com>:
> 
> 
>Привет,  @sibmaker, спасибо за отчет. 
>Я могу воспроизвести это и в версии 4.8, но я вырезал это и в нашем выпуске Arachne.  А этого не происходит.
>Вы можете узнать больше о нашем выпуске Arachne  здесь.
>- 
>Вы получаете это, потому что вас упомянули. 
>Ответьте на это письмо напрямую,  просмотрите его на GitHub или  откажитесь от подписки . 
 
 
С уважением,
Яковлев Сергей Александрович
+7-(963)-944-07-40

fvrmr, 2021-03-05:
I can reproduce this also in 4.8, so I will discuss it with the team, but using Arachne is a work around.
Do you have more files that have this same problem?

Keep you posted!

sibmaker, 2021-03-05:
The file has nothing to do with it.  This problem manifests itself on all thin-walled parts with wall thicknesses of 1.2, 2, 2.8 mm.  And not only for me alone, I interviewed 6 people and everyone has the same problem.  This is a bug in CURA.  Slice the 2mm tube and watch the visualization carefully.  The middle layer will be printed 2 times.  I will make a video and send you a link.
--
Отправлено из Mail.ru для Android пятница,

fvrmr, 2021-03-05:
Okay good to know! I saw it with your file, I was just wondering if it happened with more files. 
You don't need to make a video to show me, I have seen it myself with your file.

I will let you know when I have discussed this with the team.

sibmaker, 2021-03-05:
Okay!  Thanks you!  I look forward to solving the problem!  And thanks for the software, I really like CURA.  And the video will still be on YouTube.  I have my own channel for 3D printing
--
Отправлено из Mail.ru для Android пятница,

fvrmr, 2021-03-08:
Hi @sibmaker I have discussed it with the team. And we are aware of this issue and Arachne is the solution for this problem, so that's the reason that we won't fix it in 4.8. 
So please check out the Arachne release. 

Hope you understand. I will close this issue, but if you have more questions please reach out.

closed 2021-03-08 by fvrmr