Ultimaker/Cura #9357: Text box to input a specific angle of rotation for a model

GregValiant, 2021-03-03:
Currently, I can rotate a model to the snap points (45°, 90°, etc.).  With snap off, I can (very carefully) rotate a model by degrees (1°, 10°, 34°, etc).  There are times that I need a specific angle of rotation (Ex: 22.5°, 78.75°).  Would it be possible to include a text box with the Rotation Tool so a specific angle of rotation could be input?
**Affected users and/or printers**
Engineers importing multiple model files that need to be properly aligned to each other.

**Additional context**
Two decimal places would be good.  It would be nice if negative angles could be input.

Thanks,
Greg

fvrmr, 2021-03-04:
Hi @GregValiant thank you for your feature request. 
I will discuss it with the team. Keep you posted.

fieldOfView, 2021-03-04:
Duplicate of #2779

GregValiant, 2021-03-04:
I see a lot of thought has gone into this.  I will close this one.

closed 2021-03-04 by GregValiant

fvrmr, 2021-03-05:
Thanks @fieldOfView for referring the duplicate.