Ultimaker/Cura #9360: Ultimaker S5: Bed temperature depends on installed materials not on used materials.

Aton, 2021-03-03:
**Application version**
Cura 4.7.1

**Platform**
Windows 10

**Printer**
Ultimaker S5

**Reproduction steps**
- Install PLA on print head 1
- Install PC on print head 2
- only use PLA for printing (model, brim and support)

**Screenshot(s)**
n.a.

**Actual results**
The bed temperature goes up to the temperature level of PC (about 110°C) but PC is not in use!
For PLA this temperature is to hot.
Disabling the print head with PC is not possible

**Expected results**
Bed temerature must depend on used Material (print head) when only 1 print head is used.

**Project file**

**Log file**

**Additional information**

fvrmr, 2021-03-04:
Hi @Aton thank you for your bug report.

You're saying that you can't disable print head 2. But I am able to do this, and then the bed temperature is 60 degrees. 
[image]
Just wondering why it is not possible to do this.

Aton, 2021-03-04:
Thank You fvrmr,
Disable is really possible. My mistake.
But also without disabling, the printer sould select the temperature of the used material if there is only one in use.

fvrmr, 2021-03-05:
I totally understand that it doesn't make sense. 
But Cura can't see which extruder is used for what parts, and picks the highest temperature.
I will reopen this and discuss this with the team.

fvrmr, 2021-03-08:
Hi @Aton I have discussed it with the team and it was already a know issue, so there is a ticket on our backlog of this issue.
Developers see CURA-7225.

GregValiant, 2024-11-19:
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

Aton, 2024-11-20:
The ticket can be closed. I have tested it in Version 5.9. Here are my findings:
Cura always automatically suggests the temperature for the material with the highest bed temperature.
The problem was solved by the option of manually overriding the bed temperature. Cura still does not automatically detect when only one material is used and both extruders are activated, although this could actually be easily determined when setting the job materials.

closed 2024-11-20 by Aton