Ultimaker/Cura #9361: Zoom to mouse position (Mac)

seaniepie, 2021-03-04:
**Application version**
4.8

**Platform**
Mac, BigSur, MBP 16inch

**Printer**
CP-01

**Reproduction steps**
1. Turn on Zoom Towards Mouse Direction in prefs
2. Use two fingers on trackpad and zoom. It zooms centre if the mouse pointer is bottom right. The variable values are effectively x2 of mouse loc.

**Screenshot(s)**
na

**Actual results**
zoom is off centre

**Expected results**
zoom should be centred on the mouse current location

**Project file**
na

**Log file**
na

**Additional information**
na

fvrmr, 2021-03-04:
Hi @seaniepie could you share your log file? 
You can find that here: `$USER/Library/Application Support/cura/<Cura version>/cura.log`

Ghostkeeper, 2021-03-11:
Duplicate of https://github.com/Ultimaker/Cura/issues/6775.

closed 2021-03-11 by Ghostkeeper