Ultimaker/Cura #9363: When origin at center checked, Y coordinates in preview screen and Gcode differs

onrer, 2021-03-04:
Hello guys

Please check the issue written below. 
Cura preview screen view and Gcode output for Y axis coordinates different. (+ and - reversed)
When i move object to Y-20 slice and export Gcode object coordinates plus(+) not minus(-)

This fix could affect all users and I'm sure you guys tons of stuff to make, I'm not asking for fix. 
I'm just asking could you please point me best place to fix in Cura source or json config files?

(pls, I'm not looking for workaround how to do it Gcode or fw ect.)

I appreciate your time. 
Thank you!

**Application version**
cura 4.8.0

**Platform**
mac

**Printer**
custom printer

**Reproduction steps**
1. open printer settings set origin at center
2. in prepare screen add STL file
3. change object y axis position, by entering Y:-20
4. slice and export Gcode
5. check the generated Gcode

**Screenshot(s)**
[image]

**Actual results**
y axis coordinates of object has plus(+) coordinates

**Expected results**
y axis coordinates of object should have minus(-) coordinates

**Additional information**
my printer center is 0,0

onrer, 2021-03-04:
pls check the issue here. Its looks like, object Y axis and bed y axis are opposite.

https://community.ultimaker.com/topic/36211-cura-480-y-axis-coordinates-difference-in-software-screen-and-gcode/
( I asked the question. )

GregValiant, 2021-03-04:
I'd like to know also.  The "Cura" origin indicates that +Y is at the top of the screen.  The "part" origin indicates that +Y is at the bottom of the screen.  Why are they different?  I don't know what effect it has on anything other than display and part location as noted in the move textbox, but it is confusing.
[image]

gr5, 2021-03-04:
@onrer - did you use the mirror/flip axis tool on the left side of the screen at all?  The 4th tool down?

gr5, 2021-03-04:
I don't see this as a bug - the coordinate system for printers just happens to be backwards from the coordinate system of many graphics interfaces.  With graphics interface software, typically 0,0 is the top left corner of a window or screen.  But when you translate to gcode you have to translate it into the coordinate system of what a printer typically uses which is: positive values for x,y,z are right,back of printer, up (respectively).

onrer, 2021-03-04:
I think, its not about how graphics interface software written or implemented in computer software's. Depends on how you implement the coordinate system on GUI screen. 

And it's not about what usually printer does or ect, you cant judge this problem with that.

There is an option for "origin at center" and its not matching the GUI and gcode. thats the problem.
Which i'm not asking for fix. As I said, I'm just asking you if you could point me the best place to fix in Cura json config files or Cura source code.

fvrmr, 2021-03-04:
Hi @onrer thank you, but this is a duplicate of #2152 
It is expected behaviour at the moment, like @gr5 mentioned. 
Our front-end uses another coordinate system than the g-code is using. The reason for that is because g-code coordinates can be different per printer.

I will close this issue. But feel free to discuss this in the other thread.

closed 2021-03-04 by fvrmr

GregValiant, 2021-03-04:
As I think about it, I can't call it a bug either.  The Coordinate system of the part is simply wrong.  Positive Y is up from the origin in every coordinate system I'm aware of.  I have in my lap Machinery's Handbook (1984 edition).  On the desk in front of me is Basic Technical Drawing (1956 edition).  It is wrong in both of them.  I think it was wrong going back to Archimedes and Euclid.

There are two coordinate systems being employed in Cura and they don't agree with each other.  The Build Plate CS is what we expect to see.  It also provides the basis for what we see in a gcode file.  That remains true with the Origin At Center.  As it relates to the Part CS - the gcode would be wrong but it isn't.   Therefore, the part UCS is wrong.  What does it affect beyond translation and the location displayed in the Translation tool dialog?

gr5, 2021-03-04:
Interestingly, some Ultimaker employees consider this a bug.  In particular the cura coordinate system is used for the move tool only but the gcode coordinate system is used for placement of the tower, the purge point, and something else - see isssue #9346 for more details.

onrer, 2021-03-04:
@GregValiant ". The Coordinate system of the part is simply wrong" 
100% totally agree that. Even object sits on the center or not, its same. The problem is in the coordinate system of the part.

I also think that, when the part coordinate system implemented they just designed  giving + position moves object up otherwise down. And then some printer came up with origin at center, and they add quick fix for it.

part coordinate system has relative coordinates not absolute. When origin at center selected also same. That's why i confused.

I come up with 2 setting options ideas.  (regardless of setting origin at center or not)
1. Absolute part coordinates for the part
2. just reverse y axis 

I will start checking source code and see what I can find. 

@gr5 Thanks for the info. It makes things clearer.

@fvrmr I understand. I respect the decision. I believe there are people there who think its confusing or wrong.
I want to make a PR for this, Could you please give me a hint about where to start looking? 
Thank you.

fvrmr, 2021-03-09:
Hi @onrer you could take a look at the translation tool: https://github.com/Ultimaker/Uranium/tree/master/plugins/Tools/TranslateTool

GregValiant, 2021-03-09:
@fvrmr  - Is there a download link for the Translate Tool?  How would I install it?

fvrmr, 2021-03-09:
The link I posted is the source code of Uranium and that is already a part of Cura. 
So if you want to make a PR you could take a look at the code here and see if you can make changes. 

But please keep in mind that I'm not sure if we would accept this PR.

Ghostkeeper, 2021-03-11:
If you only need to make changes to the TranslateTool, you can make the changes in an existing installation. Like, Program Files on Windows. There's a plug-ins folder that contains Python code for the plug-ins and the TranslateTool is one of them. You can simply edit the Python code in there.

onrer, 2021-03-12:
@Ghostkeeper Thank you for the information. I will check it out.
I' planing to add a setting for Absolute coordinates for the part. 
Actually I don't know how much I can achieve in the TranslateTool. 
In TranslateTool maybe all I can do in change the y direction of part. I don't know yet.

onrer, 2021-03-13:
[image]
I could able to change part arrows directions. Next I will try to change y coordinate direction.
plugins/Tools/TranslateTool/TranslateToolHandle.py

onrer, 2021-03-13:
[image]
I could able to change part y axis coordinate.

I attached the files here for who wants to fix it, Until i came up with good setting option and make PR

plugins/Tools/TranslateTool/TranslateToolHandle.py
plugins/Tools/TranslateTool/TranslateTool.py

[y axis fix.zip](https://github.com/Ultimaker/Cura/files/6133910/y.axis.fix.zip)

GregValiant, 2021-03-13:
@onrer  You sir are the man (or woman, whichever applies).
For a year I'd move parts around on the Cura build plate and then look at the coordinates, and wonder what the heck was going on.  I am really annoyed with myself that I never noticed that the "Y" of the part was mirrored.  It was right there in front of me the whole time.
Thank you very much.

onrer, 2021-03-13:
@GregValiant no problem at all.
@fvrmr and @Ghostkeeper Thank you for pointing out the right direction.

Ghostkeeper, 2021-03-15:
It's not difficult to change in the code. The difficulty comes from the usability of the coordinate textboxes in the Move tool. As explained in the original ticket that this is a duplicate of (https://github.com/Ultimaker/Cura/issues/2152), there are two problems with this change:

* The coordinate system will change between printers.
* It's unknown where the middle is, making it harder to place objects in the centre, which is a more common use case than placing them in the bottom left corner. You'd need to know how wide the printer is and do a calculation in your head.

gr5, 2021-03-15:
But there could  be a checkbox which lets you switch between the current mode and absolute mode.  And when you change the checkbox it fills in the absolute positions automatically (similar to the scale tool).

onrer, 2021-03-16:
@Ghostkeeper I also think so, That's why I ask your opinions.

>>The coordinate system will change between printers
you mean there are printers Origin at center and bottom left origin right? are there others?
even if there are other types all printers has settings file and defined in setting files right? otherwise gcode can't be correct right?so  it can be done.

i believe confusing comes from people using cura have different needs and backgrounds.

some people only print at center so it doesn't bother them what part coordinate is. seeing 0,0 could be better.
making this setting disable by default, not effect those users.

some people want printer and build plates coordinates same so when they move printer head somewhere will know it will match build plate. Easy to work with big plates. (more professionals)

>>It's unknown where the middle is, making it harder to place objects in the center, which is a more common use case than placing them in the bottom left corner. You'd need to know how wide the printer is and do a calculation in your head.

we have printer settings we know the plate size so it will be xmid=size/2 right? 
as I assumed before,  all printers has settings file and defined in setting files right? so no problem.

as I understand putting "Absolute position" check box right of the "Lock model" not good.
its better to put in printer settings.

i made many assumption, please correct me if I'm wrong.

Thank you for your time.

Ghostkeeper, 2021-03-23:
> you mean there are printers Origin at center and bottom left origin right? are there others?

There are others (I believe Creality printers allow printing slightly into the negative coordinates) but Cura doesn't support that so these two cases are I think the only ones.

> we have printer settings we know the plate size so it will be xmid=size/2 right?
as I assumed before, all printers has settings file and defined in setting files right? so no problem.

Yes, Cura knows how big the printer is and can easily place something in the centre even if that centre would not be 0,0. But the user wants to do this manually as well.

gr5, 2021-03-23:
There are more.  Someone on the forum, in the last 2 weeks, complained because they say 0,0 is in the far (rear) right corner and positive values head towards the front left corner.

It's not a huge deal as it just prints everything rotated 180 degrees on the build plate.  They asked to be able to fix the faded grayed out graphics that shows the build plate - they want that rotated 180 degrees to match what happens when they print.  I didn't respond to the post but can find it if you want.

Ghostkeeper, 2021-03-23:
If it's just rotating the model by 180 degrees we can do that for them (paying attention to maybe change the origin so that the build plate is still aligned with the build volume). If it's a bit more nuanced we'd probably need someone with the printer to make the change for us to make sure we're not changing it wrongly again.