Ultimaker/Cura #9366: Print only desired layers.

jeq1, 2021-03-04:
I often will have some designs, where i can see with bare eyes that there might be problem.
If there was option to save only selected layers to file, then i could print those layers and see if those problems needs attention before printing whole job. 

Printing only desired layers whould save lot of time and frustation.

GregValiant, 2021-03-06:
If you lower your model into the print bed, you get rid of the lower layers.  For the upper layers you can put a Support Blocker configured as a cutting mesh and cover the top of the model.  You can also delete those upper layers from the Gcode file or abort the print at the layer of your choice.

Ghostkeeper, 2021-03-11:
This feature is typically the sort of technical manual work that we try to prevent with Cura though.

closed 2021-03-18 by no-response[bot]