Ultimaker/Cura #9368: Persist visibility of per-model settings

Nesciosquid, 2021-03-04:
### Is your feature request related to a problem?

I frequently want to turn support off for particular models when printing multiple parts at once. It's awesome that we can do that through the per-model settings UI, but I have to manually configure what settings are visible for each model before I can uncheck "Generate Support," which is tiresome.

### Describe the solution you'd like

I would much prefer if the visibility selections for per-model parameters were "global" -- that is, following the same pattern of the advanced print settings UI, where once you set a parameter as "visible" it's persisted for all future prints.

That is, instead of setting parameter visibility per-model, the visibility settings would apply to all models, and the "Select settings" button would open a shared per-model settings visibility selection that's the same for all models.

You'd have to decide whether "invisible" settings have their override values disabled, or if they're still active even though they're not visible (similar to the way the print UI works). 

### Describe alternatives you've considered

You could expose some kind of config file or UI specifically for the purpose of selecting "default" settings visibilities that any future model will start with, which might be cleaner, but more of a "power user" approach.

### Affected users and/or printers

All users who use per-model parameter overrides.

fvrmr, 2021-03-05:
Hi @Nesciosquid thank you for your feature request.
I will discuss it with the team. Keep you posted.

Nesciosquid, 2021-03-05:
Thanks!

Even if y'all don't have the time to update the existing behavior or UI, it would be super helpful to support adding some kind of config file that you can add to the Cura directory to tell it which parameters to include by default when you turn per-model settings on, so that power users can set it up manually. :)

fvrmr, 2021-03-08:
Hi @Nesciosquid I have discussed it with the team and we have decided to defer this feature request. 
But we do believe that this has value but we don't see this happening any soon. 

Hope you understand.

Ghostkeeper, 2021-03-15:
It was discussed that we'd need to change more things for this feature. Because as it works now, adding the setting in the per-object settings panel locks the value of the setting in place for that model. So if you add the Infill Density setting to an object, then change the Infill Density in the normal/global settings list, the infill density will not change for the object you added it for. So if these per-object settings are persistent, changing them in the normal settings list will no longer have any effect. This will be confusing.

So we'd need to think of a different workflow for per-object settings altogether for this.

fieldOfView, 2021-03-15:
Instead of persisting settings, it could be interesting to have per-model setting "sets" or "profiles" that a user could create/store/reuse.

closed 2023-07-13 by github-actions[bot]