Ultimaker/Cura #9372: Print Brim on top of Raft / Last Raft Layer as Second Material.

StuSerious, 2021-03-05:
### Is your feature request related to a problem?

Yes, on smaller models or ones that have a small contact patch with the raft, while interfacing two different materials, adhesion on the raft is not 100% optimal, leading to the print dethatching from the raft.

### Describe the solution you'd like

I'd like to suggest one of two setting, or both:

* *Print brim on top of raft*: Would allow the user the ability to control whether or not to generate the brim on top of the 
last layer of the raft, making adhesion easier and the part more stable.

* *Last raft layer as print material*: would possibly easier to implement in the code, and guarantee that the same material is interfacing on the raft. On the other hand, it could make part removal more challenging.

### Describe alternatives you've considered

I see no alternatives due to the specificity of the request but please do light my way in the comments.
IMHO, this is something farely unique that has no workarounds. (I mean.. I opened this PR mailny because UltimakerPP and UltimakerBreakaway are an absolute pain to print toghether especially when the part has a small footprint)

### Affected users and/or printers

Anyone in the professional / production space knows the importance of rafts and adhesion for reliability purposes, so I don't feel like I need to explain a lot..
I feel like this is a small improvement that could make some very tough models to print easier, from the single model all the way to a larger scale / production setting.
Furthermore, this is something UM printers would benefit from, especially since the Ultimaker Breakaway and PVA are *extremely* solid filaments that have very wide adoption in the semi-pro / production space.

I would program this myself but I'm still trying to learn while doing all the stuff I do so that's not going to happen anytime soon.

As always, thanks for your time!
~Ste

P.S.: Thank god for this new Issue format, I swear to god I was being so triggered by people half-assing issues and leaving forms empty I can't imagine @fvrmr 😆

fvrmr, 2021-03-05:
Hi @StuSerious thank you for your feature request!
Will discuss this with the team. Keep you posted.

And yes I'm very happy with the new issue format!! Let's see how it will work out! 😄

fvrmr, 2021-03-15:
Hi @StuSerious apologies for my late response. But I have discussed within the team and also with the material team. 
And we are curious what the advantage would be over just printing directly on the build plate with a brim? Especially with a glass build plate. Also raft is not turned on by default so normal users will not really benefit from this feature. 

Printing with PP and support material (PVA or BAM) would be quite hard with or without your feature request. PP on PVA is officially not support by Ultimaker and it would be a complicate solution for the materials. Also we are not sure if that can fix the adhesion problems of PP.

But if you really want to print with a raft, try to reduce the `raft gap air`

Let me know if you have more questions.

StuSerious, 2021-03-25:
Hey there @fvrmr! I beg you pardon, for some reason I had completely missed your reply!

Thanks for also taking the time to discuss this with the material team!

Here are some thoughts I have I'd like to share, I'd love an opinion and input from you and anyone else reading:
I agree fully that the main reason for the suggestion/request is very specific, so I'll clarify a bit what I meant, I apologize in the event that I wasn't clear 🙇🏼
A small correction first, the issue I was having was while printing a brim with HIPS and the main model with Ultimaker PP.

There are certain scenarios and machines (like ours at work) where you are forced to print on a raft, due to the manufacturer requiring it or the bed being fixed and therefore not levellable.
While printing on a raft with flexible or particular materials such as PP, models that have a small footprint with the bed/raft are very prone to being extremely unstable.
From my research, having the possibility of making that contact patch larger with a brim is very useful and can completely prevent this issue, therefore the original request. 

The `Raft Air Gap` I use is already set to 0, but I see where the suggestion comes from and I'll definitely try to tinker with it some more!

Furthermore, I think this feature would be useful in other scenarios as well for any dual/multiple extrusion setups where the model has a small contact patch or is proving tough to interface with another material!

Thanks as always for your time!
~Ste

Dr-Chronosphere, 2021-03-25:
@StuSerious and I were discussing his issue/feature request in the 3dprinting discord, and I was able to suggest a workaround just in case other people run into wanting the same sort of stacked brim/raft concoction. Essentially, you could add a brim in whatever CAD software you use and once putting it into Cura, add the raft, which would then be generated underneath the brim and model. It would likely be easier if Cura could generate both at once, but this serves as a workable solution for the time being.

StuSerious, 2021-03-26:
@Dr-Chronosphere I can't believe I hadn't thought of that myself! Thanks again for the effective workaround idea!

StuSerious, 2022-06-15:
hey there @fvrmr! 

are there any updates on this?

StuSerious, 2022-06-16:
No worries @nallath!

I can completely understand that this is a lower priority task!

I do however want to add to what was said above that KISSSlicer, S3D and others do have this feature, if that is any useful metrics to you.

Thank you for giving some time to this arguably insignificant thing.
It's one of those things that  you almost never need but when you do need it you want it so bad 🤣 

Thanks!
_~S²_

MariMakes, 2024-01-11:
Hey @StuSerious,

Quick update from our side 👋
We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, I stumbled upon your issue about printing rafts with two materials. 

The latest versions of Cura have had some improvements in this field to support the BASF Metal materials and MethodX printers with dual material rafts.

[image]

This week we are working on even more improvements to prevent warping of these top layers like adding Raft Walls (https://github.com/Ultimaker/Cura/pull/17946) and different Raft Margins (https://github.com/Ultimaker/Cura/pull/17945). We plan to include these settings in the upcoming Cura release. 

I'll close this ticket since there is a solution in place.
Thanks for your report. Together we can make Cura better 😊

closed 2024-01-11 by MariMakes

StuSerious, 2024-01-11:
Hey there @MariMakes!

At a high level the implementation seems to cover my original request!
I'm yet to test it, but i think it'll work just as other slicers do.

Thanks for closing this!
~S²