Ultimaker/Cura #9373: Retraction right at start

FlailAway, 2021-03-05:
### Application Version

Arachne_no_build_number_in_About

### Platform

Mint 20.1

### Printer

Custom

### Reproduction steps

Slice
Save
Open output GCode

### Actual results

;; LOOK NEAR THE BOTTOM OF THIS START CODE FOR THE ISSUE//QUESTION
;FLAVOR:Marlin
;TIME:3527
;Filament used: 8.86869m
;Layer height: 0.3
;MINX:81.583
;MINY:101.559
;MINZ:0.3
;MAXX:208.508
;MAXY:188.712
;MAXZ:12.9
;Generated with Cura_SteamEngine 99.9.0-Arachne_engine_alpha
M82 ;absolute extrusion mode
G91 ; relative  // CURA - MY START-START
G1 Z10 F3000 ; go up, wait to prevent cold nozzle blob
G90 ; absolute
M420 S1 Z0.00  ; restore bed leveling
M107 S0 ; fan off
M104 S215 ; set hotend
M140 S55 ; set bed
M109 S215 ; wait for hotend
M190 S55 ; wait for bed
G92 E0 ; zero extruder
G28 ; home all axes  // CURA - MY START-STOP
G92 E0
G92 E0
G1 F1500 E-6.5                      <<<<<<<<<<<<<<   WHY?  WHY?  WHY?  WHY?  WHY?  WHY?  WHY?  WHY?  WHY?
;LAYER_COUNT:43
;LAYER:0
M107

### Expected results

Why is there a retraction right at the start?

If you MUST have it, can we have an option to set the amount for :"Initial Retraction," or something so I can set it to ZERO!

I have to waste extra Skirt-lines to bring the Filament that 6.5mm back to the nozzle.

fvrmr, 2021-03-08:
Hi @FlailAway thank you for your report. 
You could change the `retraction distance` let me know if that works.

If that doesn't do the job, could you share your project file? (file - save project)

Ghostkeeper, 2021-03-15:
It should also unretract when it arrives at the skirt or the model, so introducing extra skirt lines shouldn't be necessary.

GregValiant, 2021-03-17:
This came up on the Ultimaker Forum [https://community.ultimaker.com/topic/24847-extra-retraction-at-start/](url) last fall.
I had seen that my skirts and brims didn't start right away and so a string of beads would get pulled across the area where the print would be and the string would often end up in the print.
I think the problem is oozing because the numbers do add up as @Ghostkeeper says.  In your case there is a 6.5 retraction and a 6.5 prime but the skirt doesn't start.  I my case it looks like this:
G1 F2700 E-5
G1 X0.1 Y20
G1 Z0.3 F3000
G1 F2700 E0
I use the Search and Replace post-processing plugin and my search criteria is "G1 F2700 E-5" and the replace string is "G92 E-5".  Instead of pulling the filament back from the nozzle it sets the location to -5.  The result is a decent prime when the E0 comes up a couple of lines later.  There is sometimes a small blob - but I don't pull a string into the footprint of the model.  The downside is that if I change the retraction distance or the retract speed I have to change the Search criteria to match.

closed 2021-03-22 by no-response[bot]