Ultimaker/Cura #9374: Gradual Infill steps improvement

AHS87, 2021-03-05:
### Is your feature request related to a problem?

When using the infill steps cura draws some lines on open space

### Describe the solution you'd like

To add an option that forces the new layer to begin on a place where there is material on the previous layer

### Describe alternatives you've considered

A solution could be to add an option that forces the new step to travel the full length of the previous step for a certain amount of layers.

### Affected users and/or printers

Everyone printing something with an irregular top layer who wants to save some material in the process

This is how it is working
[image]

This is what I propose
[image]

fvrmr, 2021-03-08:
Hi @AHS87 thank you for your feature request.
I will discuss it with the team. Keep you posted.

fvrmr, 2021-03-09:
Hi @AHS87 try increasing your step size. 
And if that is not helping, what is the impact on your print? 
Because if you look at layers above this layer you will see that it will connect.

AHS87, 2021-03-11:
Depending on the complexity of the top layer sometimes the steps take place on empty space, I think having some layers traveling the full length of the previous step will allow for complex surfaces and shorter steps. Sometimes it can be solved by taking long steps but this uses more material.

AHS87, 2021-03-11:
This is only an idea to improve this amazing feature allowing for prints with less material when prototyping or for prints that don't need strong infill, I hope it is taken into account for a future version.

Ghostkeeper, 2021-03-15:
In my mind, this is the same sort of problem as https://github.com/Ultimaker/Cura/issues/8983 . Improvements can certainly be made for certain infill patterns. But the same improvements don't apply to concentric infill, Cross infill, Gyroid, etc.

AHS87, 2021-03-19:
You are right, this will help only with some patterns. The idea is to be able to do prototypes saving material and time by using extremely low infill and allowing for complex top layers with a relative good quality.

fvrmr, 2021-06-04:
I had a small discussion with a colleague about this issue because we didn't reach a conclusion and got a new issue related to this, but we decided to defer this issue.
The reason why is that it would build up eventually to be supported and like @Ghostkeeper mentioned above it is made for certain infill patterns.

dsvilko, 2021-08-31:
It's very unfortunate that this is closed. Printing in mid-air is never good and can often lead to problems. I really believe the infill algorithm should be modified such that it can only start and stop the printing of a line over a previously printed line of the previous layer. For objects with complex geometry gradual infill steps currently provide very little or no roof support making the option useless for anything other than simple, flat roofs.

fvrmr, 2021-09-02:
@dsvilko it is not closed, we only made the decision that we don't have the time and priority to add this ourselves now. Everybody else is always welcome to make a pull request for this.

Ghostkeeper, 2021-09-10:
Or we could see new opportunities in this later if Ultimaker's priorities shift or if new reasons arise why this needs to be implemented. Repeating the old arguments won't make much of a difference, but if you can argue that there is a good reason why this should get higher priority, we do listen.

Currently, our experience is that it's enough for an infill line to attach to the wall on one side. The line will stick there and provide a bit more support for the infill line on the next layer, so that it can provide more support on the next layer and so on. In practice the infill is fully supported after a few layers. Attaching the infill lines on both sides could reduce the number of layers it takes for this to be properly supported, at the cost of using some extra material on those layers. We estimate that it'll save very little material and time in the end.
Other disadvantages of this approach are:
* Increased slicing time with grid, lines, triangles, tri-hexagon, cubic, octet and quarter cubic patterns to figure out where the next line intersects.
* Can't work with Concentric.
* Can't work with zigzag?
* Can't work with lines?
* It's unsure how this would work with the Cross, Cross 3D and Gyroid infill patterns, but we could be looking at simply generating the infill pattern at two densities and doing line-by-line intersection which is very expensive.
* The increased slicing time could expand beyond the time saved in printing, or at least beyond equal time value assuming operator time is 10 times as valuable as printer time.

Those are the reasons why we don't consider this to be worth our time investment at the moment, so we didn't add it to our priority list. If anything happens to change those reasons, it can certainly be reconsidered.

SxeyChick, 2022-10-04:
I am affirming that this issue does affect simple prints to full failure with almost every single infill. 

Here is a simple box with an embossed word on in. A simple a common thing to print. 

You can see that for every single infill (other than cubesub, cross, and cross3d) the gradual infill under the word is either fully unsupported, or 90% unsupported, both would result in full failed of the word to print. 

This issue needs to be fixed. 

My suggestion! The gradual infill could act like lightning infill off of the other infill or the walls of the print if it’s not supported on both ends (or at all).

[image]
[image]

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

SxeyChick, 2022-10-04:
Or you could have a mix of “Infill Support” and “Gradual Infill” for these common “island” areas: do gradual infill for every part of the print that goes bottom to top without dipping back down at all, and do infill support for the islands.

SxeyChick, 2022-10-04:
If you’re wondering, this is printed with a 1.2 nozzle at .9 wall width, 2 walls. With that thickness of wall, I don’t need much infill for it to be strong. I was going to go with zigzag or lines, both of which have enough to give the object some weight and structure. But the embossed word would fully fail with any of these infill.

canz, 2023-10-05:
i believe adding something like 'Bridge Sparse Infill Max Density' called 'Gradual Infill Sparse Max Density' and a corresponding infill speed\fan speed  would fix the issue for gradual Infill patterns trying to span a gap to raise the density of the infill