Ultimaker/Cura #9378: How to get OutputDevices' saved file name and location in plugins.

ChristopherHoffman, 2021-03-06:
**Application version**
4.8.0 +

**Question**
I've been developing a plugin for saving 3d print settings to a website, [3D Print Log Uploader Plugin](https://github.com/ChristopherHoffman/3d-print-log-cura-plugin), and I'd like to start recording the filename of the exported gcode file, and possibly creating a hash of the gcode file for use with other integrations.
(ie, saving settings for a print from Cura, and updating that same record from Octoprint by using the filename or hash to link them).

Is there a way for a plugin to get the saved file name and location after a user has saved the gcode?

I've looked through the OutputDevice methods, and the Application's PrintInformation methods, but I can't find a way to get the filename that the user actually exported as. The OutputDevice creates the Job with all that information during save, but only has a [local reference to that job](https://github.com/Ultimaker/Cura/blob/28a0604951b4789e318ff860cffec9433d750db3/plugins/RemovableDriveOutputDevice/RemovableDriveOutputDevice.py#L145-L151), and when writing is finished/successful the OutputDevice just returns itself and not the job information.

nallath, 2021-03-09:
I don't see a way to do that with the current code. Changes would have to be made to the existing Cura code in order to support that. 

It shouldn't be too hard to add it. If you feel like it, you could have a try at making a pull request for it.

Ghostkeeper, 2021-03-15:
It'd depend very much on the output device used.
* The Removable Drive output device indeed thinks up its own name based on the job name and saves it there.
* The Local File output device thinks up a name based on the job name and suggests that as the default name to save to, but allows the user to change the name.
* The USB printing plug-in doesn't think of any name at all. It just sends it over serial port.
* The UM3NetworkPrinting doesn't save the file anywhere either, but makes a POST request to the printer firmware or the Google server that stores prints for the cloud printing features. This uses the job name but not a real file name.

So if this is to be added it couldn't be common to all output devices. Just specific to certain plug-ins.

nallath, 2021-03-16:
The UM3NetworkPlugin does send a file name with it. It even sends an extension as far as I remember.

closed 2021-05-03 by Ghostkeeper