Ultimaker/Cura #9379: An easy way to emboss recylcing codes on to 3D models

traverseda, 2021-03-07:
### Is your feature request related to a problem?

Climate change I guess ;p

### Describe the solution you'd like

A way to emboss a recycle code on to a part, from inside Cura. Generic text would also be nice, there are times when embossing the file name would have been very useful to me.

### Describe alternatives you've considered

Manually embossing a recycle code using OpenScad or other CAD tools

### Affected users and/or printers

3D printers are becoming more mainstream, I recognize that this would be a difficult feature to add as it would be a feature you'd find in a full featured CAD program. None the less I think it would be good to make this as easy as possible for end users, so that it actually gets used. Very rarely is there any kind of requirement to add a recycling code, but what that means is you end up with a bunch of un-recycled plastic.

I doubt most users are able to bring a part into a CAD program in order to add a recycle code, especially with more complicated surfaces.

It's a complicated feature, and I think most things won't end up recycled. But at the kind of scale Cura is used in if even 5 percent of the parts printed get recycled that's going to end up being millions of tons of CO2 over the lifetime of the project, well over any individuals lifetime carbon footprints. It's the kind of things I think could have a pretty big impact on the entire world.

fvrmr, 2021-03-08:
Hi @traverseda thanks for your feature request. 
I will discuss it with the team. Keep you posted.

traverseda, 2021-03-08:
Good luck! Hopefully it's not as difficult to implement as I imagine it would be

fvrmr, 2021-03-09:
I have discussed it with the team, and there was already a similar ticket on our backlog. 
Developers see CURA-7389

traverseda, 2021-03-16:
@Ghostkeeper I was imagining it as more of a stamp tool. The user can find their own surface and sort of "stamp" the information on to their print, likely as an emboss.

Ideally it would support simple curved surfaces as well as flat surfaces. That's potentially quite challenging though, as it would be more or less CSG modeling? Maybe there's a simpler way to apply a height map to the surface of a model without depending on full CSG capability?

It's definitely technically challenging.

GregValiant, 2021-03-17:
@traverseda, Have you tried the Microsoft 3d Builder "Emboss Tool"?  It works on compound-curved surfaces as well as planes and the letters can be raised (+Z) or relieved (-Z).

[image]

traverseda, 2021-03-17:
@GregValiant 'fraid not, I pretty much only use open source stuff. Looks pretty much perfect though.

GregValiant, 2021-03-17:
I hold Microsoft in low esteem.  As I've played with 3d Builder a little bit, I'm impressed with some of the tools it has.  It is pretty good at repairing STL's too.

Ghostkeeper, 2021-03-23:
We quite strictly keep model editing out of scope for Cura. Cura is not modelling software, and it should not scope-creep into Cura's capabilities. That would greatly increase the workload for our developers. So I think that embossing a code like that is not going to be accepted in Cura. CSG is indeed very complex since the input model for Cura needs to be a surface mesh. This tends to be very unreliable if the input mesh is even slightly inconsistent (e.g. if the normals are flipped in some parts).

Something like Horizontal Expansion only makes it into the scope for Cura because it can be used to compensate for printing inaccuracies.

Microsoft has no doubt created some incredible tools. I like to credit that more to the skilled developers working on that tool than to the organisation as a whole though ;) If you're looking for an open source alternative, Blender can do similar things, although it's a bit [harder to use](https://blender.stackexchange.com/questions/44758/emboss-text-onto-curved-surface).

traverseda, 2021-03-23:
Yeah, that's not surprising. I totally understand why that would be the case.