Ultimaker/Cura #9381: Retraction setting causes extreme long printing

jatson, 2021-03-08:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Voron 2.4 300mm

### Reproduction steps

Generated STL with and without retraction setting turned on.
Nothing else changed either on the printer or in cura settings

### Actual results

- Without retraction the print is running fine. Printing time around 30mins with the STL I need
- With retraction set after drawing the skirt the printing time jumps to some extreme big values (for eg. 43 hours) and the nozzle stays in place for several sec after every movement

### Expected results

Printing time remains the same (or nearly the same).

Profile and generated STL attached here.
[profile_stl.zip](https://github.com/Ultimaker/Cura/files/6101476/profile_stl.zip)

Without retraction screen from printing times:
[image]

With retraction screen from printing times:
[image]

Klipper used on the printer.
Starting code:
[gcode_macro PRINT_START]
gcode:
    G1 Z20 F3000                   ; move nozzle away from bed
    BED_MESH_PROFILE LOAD=HOT ;Load hot bed mesh profile
    variable_startx: 2
    variable_starty: 0
    variable_layer_height: 0.24
    gcode:
    G0 X2 Y10 Z0.24
    G92 E0 ;zero the extruded length
    G0 Z0.15 F500 ; Lower nozzle
    G0 X52 E8 F600;  Extrude to prime nozzle
    G0 Z0.3; Lift before print starts to leave ooze behind
    G92 E0 ;zero the extruded length

fvrmr, 2021-03-09:
Hi @jatson thank you for your report. 
Could you please add an project file? (file - save project) I can't import your curaprofile in Cura. So I can't really reproduce what is happening. 
Thank you in advance.

jatson, 2021-03-12:
Hi!

Thanks for checking it. 
The file attached here.

[V300_gear.zip](https://github.com/Ultimaker/Cura/files/6128883/V300_gear.zip)

jatson, 2021-03-12:
I think I found the problem and it is not the Cura.
The Klipper itself caused it due to some flow calculation. 
Thanks for taking care anyway and sorry for the false alarm

closed 2021-03-12 by jatson

fvrmr, 2021-03-12:
Ah good to know you've found your problem!