Ultimaker/Cura #9383: Can't print via Cloud

ianpaschal, 2021-03-10:
### Application Version

4.8.0

### Platform

MacOS 11.2.1

### Printer

Ultimaker S3

### Reproduction steps

Load up Cura, sign in to DF, load my STL, slice, and attempt to print via Cloud.

### Actual results

Nothing happens when clicking the Print via Cloud button.

### Expected results

Progress bar appears and "Print sent via cloud successfully" or some such.

```
2021-03-10 09:00:51,240 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [61]: Refreshing the access token.
2021-03-10 09:00:51,623 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [250]: Attempting to store the auth data
2021-03-10 09:01:03,243 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-03-10 09:05:11,136 - DEBUG - [MainThread] UFPWriter.UFPWriter._createSnapshot [45]: Creating thumbnail image...
2021-03-10 09:05:11,146 - DEBUG - [MainThread] UM.Scene.Camera._preferencesChanged [255]: Changing perspective mode to perspective.
2021-03-10 09:05:11,153 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/overhang.shader]...
2021-03-10 09:05:11,162 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/transparent_object.shader]...
2021-03-10 09:05:11,170 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/striped.shader]...
2021-03-10 09:05:11,718 - DEBUG - [Thread-4] UM.FileHandler.WriteFileJob.run [80]: Writing file took 0.6198079586029053 seconds
2021-03-10 09:05:11,802 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestError [374]: request[fbe7f41b][put][PyQt5.QtCore.QUrl('https://api.ultimaker.com/cura/v1/jobs/upload')][timeout=10][b'{"data": {'...] got an error 302, Error transferring https://api.ultimaker.com/cura/v1/jobs/upload - server replied: Bad Request
2021-03-10 09:05:12,469 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [284]: SliceInfo sent successfully
2021-03-10 09:05:28,786 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Applications/Ultimaker Cura.app/Contents/Resources/qml/QtQuick/Controls/Button.qml:99: TypeError: Cannot read property of null
2021-03-10 09:05:28,793 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/Tools/RotateTool/RotateTool.qml:25: TypeError: Cannot read property 'tool_button' of null
2021-03-10 09:05:28,798 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Applications/Ultimaker Cura.app/Contents/Resources/qml/QtQuick/Controls/Button.qml:99: TypeError: Cannot read property of null
2021-03-10 09:05:28,803 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/Tools/RotateTool/RotateTool.qml:43: TypeError: Cannot read property 'tool_button' of null
2021-03-10 09:05:28,809 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Applications/Ultimaker Cura.app/Contents/Resources/qml/QtQuick/Controls/Button.qml:99: TypeError: Cannot read property of null
2021-03-10 09:05:28,816 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/Tools/RotateTool/RotateTool.qml:64: TypeError: Cannot read property 'tool_button' of null
2021-03-10 09:05:28,821 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/Tools/RotateTool/RotateTool.qml:83: TypeError: Cannot read property 'checkbox' of null
2021-03-10 09:05:29,093 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [315]: Attempting to kill the engine process
2021-03-10 09:05:29,097 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [231]: Previous socket existed. Closing that first.
2021-03-10 09:05:29,114 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [188]: Socket state changed to Listening
2021-03-10 09:05:29,154 - INFO - [MainThread] UM.Backend.Backend.startEngine [90]: Started engine process: /Applications/Ultimaker Cura.app/Contents/MacOS/CuraEngine
2021-03-10 09:05:29,161 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [110]: [Backend] Calling engine with: ['/Applications/Ultimaker Cura.app/Contents/MacOS/CuraEngine', 'connect', '127.0.0.1:49674', '']
2021-03-10 09:05:29,170 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2021-03-10 09:05:29,170 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [192]: Socket state changed to Connected
2021-03-10 09:05:29,181 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [181]: Backend connected on port 49674
2021-03-10 09:05:29,181 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Cura_SteamEngine version 4.8.0
2021-03-10 09:05:29,185 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Copyright (C) 2020 Ultimaker
2021-03-10 09:05:29,191 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2021-03-10 09:05:29,200 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is free software: you can redistribute it and/or modify
2021-03-10 09:05:29,210 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] it under the terms of the GNU Affero General Public License as published by
2021-03-10 09:05:29,218 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] the Free Software Foundation, either version 3 of the License, or
2021-03-10 09:05:29,228 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] (at your option) any later version.
2021-03-10 09:05:29,239 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2021-03-10 09:05:29,251 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is distributed in the hope that it will be useful,
2021-03-10 09:05:29,263 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2021-03-10 09:05:29,275 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'Dirt Road (Double, Edge) V2 v1.stl'>,rot.=Q<-0.0,-0.0,-0.7071067811865475,w=0.7071067811865476>), took 40ms
2021-03-10 09:05:29,275 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2021-03-10 09:05:29,285 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] GNU Affero General Public License for more details.
2021-03-10 09:05:29,304 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2021-03-10 09:05:29,313 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] You should have received a copy of the GNU Affero General Public License
2021-03-10 09:05:29,325 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2021-03-10 09:05:29,352 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: PlatformPhysicsOp.(trans.=<0.0,6.25,0.0>), took 2ms
2021-03-10 09:05:30,762 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [51]: No more on-app-exit callbacks to process. Tell the app to exit.
2021-03-10 09:05:30,767 - INFO - [MainThread] cura.CuraApplication.closeApplication [603]: Close application
[... 390 lines cut ...]
2021-03-10 09:24:10,812 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [898]: Backend quit with return code 0. Resetting process and socket.
2021-03-10 09:24:12,115 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-03-10 09:24:12,226 - DEBUG - [MainThread] UFPWriter.UFPWriter._createSnapshot [45]: Creating thumbnail image...
2021-03-10 09:24:12,234 - DEBUG - [MainThread] UM.Scene.Camera._preferencesChanged [255]: Changing perspective mode to perspective.
[... cut ...]
2021-03-10 09:24:39,612 - DEBUG - [Thread-4] UM.FileHandler.WriteFileJob.run [80]: Writing file took 1.1962449550628662 seconds
2021-03-10 09:24:39,676 - DEBUG - [MainThread] UM3NetworkPrinting.src.UltimakerNetworkedPrinterOutputDevice._reconnectForActiveMachine [245]: Reconnecting output device after timeout.
2021-03-10 09:24:40,043 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [284]: SliceInfo sent successfully
2021-03-10 09:24:48,115 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
```

[image]

nallath, 2021-03-10:
Remove the ( and ) from the name in the left corner. We made a mistake with syncing the allowed characters between Cura and the cloud. Cura does allow them, the cloud does not (and cura isn't communicating the issue back to the user)

ianpaschal, 2021-03-10:
@jellespijker Indeed, that was the first thing I tried.

@nallath Nice. That solves it. I notice comma isn't allowed either.

ianpaschal, 2021-03-10:
I'll leave it to you guys to decide if you want to keep this open or not. It sounds like it's a known issue on your backlog though.

Ghostkeeper, 2021-03-23:
It is a known issue indeed, but it's good to keep it open in a public place so that people can find the issue if they encounter it.

closed 2024-12-08 by github-actions[bot]