Ultimaker/Cura #9385: [4.8] Skin Removal Width rounding issue

Timocop, 2021-03-11:
### Application Version

4.8.0

### Platform

Windows 10 x64

### Printer

Custom FFF Printer

### Reproduction steps

1) Select "Normal" preset
2) Import CaliCat
3) No skin created at paws and face.

### Actual results

Does not create skin

### Expected results

Should create skin at those regions

[calicat.zip](https://github.com/Ultimaker/Cura/files/6122731/calicat.zip)

Does not create skin at regions when setting is on default.
[image]

When the value is slightly lower, it will generate skin in those areas.
[image]

fvrmr, 2021-03-12:
Hi @Timocop thank you for your report.
Could you add a project file? (file - save project) 
That helps me reproduce this issue. Thank you in advance.

Timocop, 2021-03-12:
This happens on a default profile preset. But here:

[calicat.zip](https://github.com/Ultimaker/Cura/files/6131861/calicat.zip)

fvrmr, 2021-03-19:
Hi @Timocop this is fixed by our Arachne engine.
Please check that out here: https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_beta

[image]

Let me know if this works for you to and feel free to reach out for more questions.

Timocop, 2021-03-22:
Yes this works very well.

closed 2021-04-08 by Ghostkeeper