Ultimaker/Cura #9387: Remove M140 S0 M107 at the end of the print

gretzke, 2021-03-11:
At the end of a print cura automatically inserts `M140 S0` `M107` and turns of the bed. As described in [this issue](https://github.com/Ultimaker/Cura/issues/6633), this can lead to the bed being turned off before the the print actually finishes. It's not possible to mitigate this issue by inserting a pause or `M400` command in the end gcode, as these commands are executed right before the end gcode.

My suggestion would be to insert an `M400` command before these commands or remove them, as they are in the default end gcode anyway or at least provide an option to disable these commands.

Ghostkeeper, 2021-03-11:
The Reddit thread this spawned from, for more information: https://www.reddit.com/r/Cura/comments/m2t1a2/bed_turns_off_before_print_finishes/

Ghostkeeper, 2021-03-11:
What if you start your end g-code with `M140 S{material_bed_temperature}`? That would immediately turn the bed back on.

I wouldn't think that the bed turning off 10 seconds too early would be a problem for any real print.

gretzke, 2021-03-11:
It’s probably not a real issue that will seriously affect most prints but here is why I think it should be removed anyway.

I don’t think most people expect it to be there, as i haven’t seen it communicated anywhere. Because the default profiles turn off the bed and fan anyway in their end gcode anyway, it’s redundant. If someone wants to keep their bed on after a print they would probably comment out the "turn off bed command" expecting their bed to stay on which would not happen. One would have to check the generated gcode file to find the problem. This isn’t beneficial to user experience in my opinion.

I’d even go this far to say that both commands should probably be replaced by the M400 command, especially when choosing Marlin as their gcode flavor. Buffering the movements is a standard feature for Marlin now so I'm probably not the only one experiencing this. Waiting for the print to finish with M400 would prevent the side effects I was seeing. Before I put M400 into my end gcode I had instances of the hotend cooling below 130°C before the print finished, which obviously ruined it.

GregValiant, 2024-11-19:
This has not been changed in the latest version of Cura (5.9.0-beta.2).  Can this be closed?  I don't see it as a bug myself but rather as "expected behavior".

closed 2024-12-08 by github-actions[bot]